RPi-Distro/firmware-nonfree #1: Source location of brcmfmac43430-sdio.bin

anadahz, 2016-08-07:
It seems that the firmware files for the BCM43430 and other wireless chipsets (brcmfmac43430-sdio.bin) are not available neither in the kernel [linux-firmware](https://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/tree/brcm) neither in the Debian package [firmware-brcm80211](https://packages.debian.org/sid/firmware-brcm80211).

It will be really nice if we could know the source location of [brcmfmac43430-sdio files](https://github.com/RPi-Distro/firmware-nonfree/commit/3f15768b68c30e09dbd5647a66fdc41dbac612b4)  so that we can submit and update the linux-firmware and the Debian package firmware-brcm80211.

XECDesign, 2016-08-07:
The files were provided directly by Broadcom. The files can be used under [this license](http://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/tree/LICENCE.broadcom_bcm43xx).
More information [here](https://github.com/raspberrypi/linux/issues/1325).

anadahz, 2016-08-07:
According to this: https://github.com/raspberrypi/linux/issues/1325#issuecomment-217358527 (from the linked issue) we wait for Broadcom to submit the wifi and bluetooth firmwares to linux-firmware.

Since the license allows redistribution (with certain limitations) can't we just submit the firmware files to linux-firmware and Debian package firmware-brcm80211 ?

XECDesign, 2016-08-08:
If someone would like to volunteer, it wouldn't hurt to raise the issue with the relevant maintainers and see what the hurdles might be.

Forceflow, 2017-01-03:
I think this is part of an issue that keeps the Raspberry Pi 3 Wifi from working when updating from jessie to stretch - https://www.raspberrypi.org/forums/viewtopic.php?t=150543&p=988948

XECDesign, 2017-01-03:
Looks like it was added upstream a few months ago:
http://git.kernel.org/cgit/linux/kernel/git/firmware/linux-firmware.git/commit/?id=c4c07a8d1128d50a5c2885ceea1abbebaa82f820

closed 2017-01-03 by XECDesign