RPi-Distro/firmware-nonfree #3: Rasbian (2019-06-20) has changes to brcmfmac43455 nvram configuration

agherzan, 2019-06-27:
Namely, `boardflags3` value is changed from "0x48200100" to "0x44200100". As far as I know this changes the optional 32khz clock input configuration. How come we don't have this change part of this repository? The changelog entry in the deb package is also not helpful in understanding this change:

```
firmware-nonfree (1:20190114-1+rpt2) buster; urgency=medium

  * Update brcmfmac43455-sdio.txt
    - Update boardflags3

 -- Serge Schneider <serge@raspberrypi.org>  Thu, 18 Apr 2019 14:35:34 +0100
```

Does this configuration only apply for RaspberryPi 4?

XECDesign, 2019-07-18:
Sorry about the delay, it should be resolved now.

agherzan, 2019-07-18:
I'm not sure what you mean by that. The PR is still open and there is no additional information about this change as far as I know. Could you please clarify?

XECDesign, 2019-07-18:
I've pushed the internal changes, which bring in the Buster updates and update the nvram patch. I'm not sure why github doesn't show a merge conflict on the PR, but the files look identical to me.

agherzan, 2019-07-18:
That is indeed really strange. Do you have any details on what that change is actually doing? Is that backwards compatible on other RaspberryPis?

XECDesign, 2019-07-18:
I don't know what it does exactly, but it is backwards compatible.

agherzan, 2019-07-18:
That is enough. Thanks for feedback.

closed 2019-07-18 by agherzan