RPi-Distro/firmware-nonfree #5: Updated wifi firmware

floion, 2019-09-27:
Hi, do you happen to have updated wifi firmware for the raspberry pi?

I see

```
strings brcmfmac43430-sdio.bin |grep -i version
wl%d: %s %s version %s FWID 01-%x
43430a1-roml/sdio-g-pool-p2p-idsup-idauth-pno-pktfilter-wowlpf-keepalive-aoe-lpc-swdiv-srfast-fuart Version: 7.45.98.38 (r674442 CY) CRC: ffb2ee5a Date: Mon 2017-10-23 03:58:00 PDT Ucode Ver: 1043.2099 FWID 01-e58d219f
```
and
```
strings brcmfmac43455-sdio.bin |grep -i version
%s: unsupported bdc protocol version (%d)
wl%d: %s %s version %s FWID 01-%x
Driver version mismatch. expected = %d, got = %d
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv Version: 7.45.154 (r684107 CY) CRC: b1f79383 Date: Tue 2018-02-27 03:18:17 PST Ucode Ver: 1043.2105 FWID 01-4fbe0b04
```

which are pretty old.

Thank you

XECDesign, 2019-09-27:
Seems to match what we're shipping in Raspbian, so it looks up to date to me.

floion, 2019-09-27:
It is on par with Raspbian, yes, but Cypress has released newer firmware: https://community.cypress.com/thread/44957

XECDesign, 2019-09-27:
@pelwell Did I drop a release somewhere?

pelwell, 2019-09-27:
No - we aren't notified of new releases, so were unaware of this. We learnt earlier this week that there is an even newer release: https://community.cypress.com/docs/DOC-17441
Within the Russian-doll of an archive there are firmwares for the 43438 and 43455. Early tests by other suggest it is better than what we are shipping now, but they need more extensive testing.

floion, 2019-09-27:
So the plan would be to wait for more testing on the release from https://community.cypress.com/docs/DOC-17441 or 7.45.173 would be stable enough to be included?

pelwell, 2019-09-30:
As far as I'm concerned the 7.45.173 has had no testing. If you've heard nothing in a week come back and nag here, but unless something unexpected happens (and it's looking good so far) we should have released the update by then.

floion, 2019-10-09:
Hi, just wanted to see how is the new firmware testing going. Is it ready to be merged? Thank you

MilhouseVH, 2019-10-09:
No issues reported so far.

pelwell, 2019-10-09:
Thanks for the quick response. It's probably time to get it into the firmware package.

pelwell, 2019-10-09:
See 4ee44af381d55bc2221cf80c1433842a48ed09e1.

XECDesign, 2019-10-09:
Updated and tested the raspbian packages. Should be public within an hour.

floion, 2019-10-09:
Thanks everybody for update.

closed 2019-10-09 by floion

tswaehn, 2019-10-11:
https://github.com/RPi-Distro/firmware-nonfree/issues/6