RPi-Distro/firmware-nonfree #6: brcmfmac43455 issues using wpa_cli with recent update

tswaehn, 2019-10-11:
I did notice issues when I use the $ wpa_cli -i wlan0 status

Since the commit 4ee44af381d55bc2221cf80c1433842a48ed09e1

the wpa_cli is no longer returning proper status feedback. Instead it complains about not beein able to find global ressource wlan0.

XECDesign, 2019-10-11:
Works for me on Raspbian. Did you also add the new clm_blob file?

tswaehn, 2019-10-12:
Yes.

I did use stretch 32bit on a rpi3b+ having the issue.

pelwell, 2019-10-14:
It works for me as well. What's in your wpa_supplicant.conf?

pelwell, 2019-10-14:
In particular, it should include a country code, e.g.:
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=GB

network={
        ssid=...
```

tswaehn, 2019-10-14:
That’s in for sure. Please note I am using stretch 32bit. 

Strange thing is that old firmware runs perfect. But new one is not working. Two identical images - just replacing the firmware changes the behaviour.

Maybe there is another thing that I am missing. Ex a newer version of wpa_supplicant

tswaehn, 2019-10-14:
BTW. The Network section is set to non existing ssid by purpose. Because I am connecting through wpa_cli dynamically.

Please confirm that connecting wpa_cli is running fine on your stretch  32bit with new firmware.

pelwell, 2019-10-14:
Do you known which Stretch image you installed, so I can match it?

tswaehn, 2019-10-14:
To save your time. Which Image and RPI Version did you use?
I can do the test here :)

pelwell, 2019-10-14:
The "Stretch" comment is the clincher. There appears to be a dependency between the firmware and the kernel version (or it's just possible it's the firmware). Installing the latest kernel with `sudo rpi-update` should fix the problem for you.

If you've updated Stretch using apt/apt-get and ended up with an old kernel and a new WiFi firmware that don't work together then we have a problem. If, however, you've just patched Stretch with the new firmware then it's reasonable to expect (now that we know) that users update both together.

XECDesign, 2019-10-14:
The wifi firmware package hasn't been updated in Stretch. So you'd only end up in the situation if you've copied it manually then.

tswaehn, 2019-10-14:
I am actually building my own stretch image and thought to have used the latest kernel and did copy the latest firmware. So following your information, i am actually not using the latest kernel then. 

Let me double check...

XECDesign, 2019-10-31:
Feel free to re-open if you can provide instructions on how to reproduce the problem starting with a clean Buster image.

closed 2019-10-31 by XECDesign

tswaehn, 2019-11-02:
Indeed - my fault: using kernel 4.14.x with latest firmware doesnt work.

Sorry for bugging.