RPi-Distro/firmware-nonfree #8: rpi4 buster bluetooth and wlan conflict

akicker, 2020-03-30:
With the update from 27.3. (updating bluetooth, ... to bluetooth (5.50-1.2\~deb10u1)) raspbian buster on rpi4 bluetooth and wlan did work well (the first time on my rpi4) !!! :+1: 
Unfortunately updates from 30.3. to bluetooth (5.50-1.2\~deb10u1+rpt1) and 11 other updates bluetooth and wlan conflicts again (BT off --> wlan ok, BT on --> very low wlan rx-rate). :-1: 
current using buster kernel 4.19.97

pelwell, 2020-03-30:
Cypress have been notified.

akicker, 2020-03-31:
Meanwhile i decided to use a **workaround** to hold blue* at 5.50-1.2~deb10u1 and run all other updates --> wlan is still working excellent 👍 
- sudo apt update -y
- sudo apt-mark hold bluez
- sudo apt-mark hold bluetooth
- sudo apt-mark hold bluez-obexd
- sudo apt-mark hold libbluetooth-dev
- sudo apt-mark hold libbluetooth3
- sudo apt full-upgrade

JsBergbau, 2020-04-01:
Just a warning for all that want to hold " 5.50-1.2~deb10u1" 
On Pi Zero W it reduces the bluetooth range dramatically. Any findings about that on PI4?

akicker, 2020-04-02:
My personal expereience (hold only on rpi4 !!!):

RPI | Model | Rev | OS | kernel | BCM | blue* | Wlan+BT
-- | -- | -- | -- | -- | -- | -- | --
3 | B | 1.2 | buster | 4.19.97 | 43430/1 | 5.50-1.2~deb10u1+rpt1 | OK
3 | B Plus | 1.3 | stretch | 4.19.66 | 4345/6 | 5.43-2+rpt2+deb9u2 | OK
4 | B | 1.1 | buster | 4.19.97 | 4345/6 | 5.43-... | NOT OK
4 | B | 1.1 | buster | 4.19.97 | 4345/6 | 5.50-1.2~deb10u1 | OK
4 | B | 1.1 | buster | 4.19.97 | 4345/6 | 5.50-1.2~deb10u1+rpt1 | NOT OK

pelwell, 2020-04-03:
I think there's an element of randomness in these results. I'm reasonably convinced that the most recent version of the 43455 firmware doesn't always play nicely with Bluetooth, but the results aren't consistent - I get laggy response from a BT keyboard with occasional connection loss on one boot, and great performance while streaming video over WiFi on the next.
N.B. The WiFi firmware is part of the firmware-80211 package. You can downgrade it with:
```
$ wget http://archive.raspberrypi.org/debian/pool/main/f/firmware-nonfree/firmware-brcm80211_20190114-1+rpt3_all.deb
$ sudo dpkg -i ./firmware-brcm80211_20190114-1+rpt3_all.deb
```
(The `./` is necessary) Yes  - rpt3 misses out rpt4, but that was later reverted.

Returning to the latest version is just `sudo apt install firmware-brcm80211`.

XECDesign, 2020-04-03:
@pelwell It looks like there are reports of the bluez package being a factor, not just the firmware.

The changes we make are the 'Add avdtp_set_priority' and hciattach patches. Any chance they could be related?

pelwell, 2020-04-03:
hciattach either downloads the firmware patch or it doesn't - it doesn't make any changes to it, so I don't believe it could be making a difference during normal use of Bluetooth. If the pure upstream version works as well then we could switch to using that.

The priority setting code is possibly a factor, since it alters the scheduling between the BT and WiFi, designed to give preference to Bluetooth audio streaming.

pelwell, 2020-04-03:
To diagnose the issue I've been rebuilding BlueZ 5.50 with and without the downstream patches, but it's been behaving itself - so I started again.

A clean installation of the latest Buster image, fully upgraded, is streaming HD video while sending audio to a headset, while two keyboards are connected. So far there have been no hiccups.

Hmmm...

akicker, 2020-04-04:
My experience downgrading the firmware to rpt3: Pi3-improoved Pi4-worsened:

RPI | Rev | OS | kernel | BCM | blue* | rpt | brcm80211_20190114-1 | Status
-- | -- | -- | -- | -- | -- | -- | -- | --
3B | 1.2 | buster | 4.19.97 | 43430/1 | 5.50-1.2~deb10u1+rpt1 | rpt1 | +rpt5 | OK
3B | 1.2 | buster | 4.19.97 | 43430/1 | 5.50-1.2~deb10u1+rpt1 | rpt1 | +rpt3 | Improved
3Bplus | 1.3 | stretch | 4.19.66 | 4345/6 | 5.43-2+rpt2+deb9u2 |   | +rpt5 | OK
4B | 1.1 | buster | 4.19.97 | 4345/6 | 5-43-... |   | +rpt5 | NOT OK
4B | 1.1 | buster | 4.19.97 | 4345/6 | 5.50-1.2~deb10u1 |   | +rpt5 | OK
4B | 1.1 | buster | 4.19.97 | 4345/6 | 5.50-1.2~deb10u1 |   | +rpt3 | Worsened
4B | 1.1 | buster | 4.19.97 | 4345/6 | 5.50-1.2~deb10u1+rpt1 | rpt1 | +rpt5 | NOT OK
4B | 1.1 | buster | 4.19.97 | 4345/6 | 5.50-1.2~deb10u1+rpt1 | rpt1 | +rpt3 | Worsened

djibux, 2020-04-04:
It seems I have the issue on a 3b+ with a fresh Raspbian install.  I have a bluetooth keyboard (Logitech K480) that I can pair properly, but it's unusable when the WiFi is active.  I tried downgrading packages to `5.43-2+rpt2+deb9u2` but the issue remains.

akicker, 2020-04-04:
If your OS is **buster** try 5.50-1.2~deb10u1 (3B+ amd 4 use the same WL/BT-chip)

zorazor, 2020-04-05:
With a downgrade to "5.50-1.2~deb10u1" my RPI4 can't connect to keyboard/mouse.

pelwell, 2020-04-07:
Connecting to a 2.4G AP I see the problem more reliably now. Cypress have also been able to reproduce the issue.

zorazor, 2020-04-08:
Thank you for the workaround hint.
I disabled 2.4Ghz WIFI on my AP and enabled 5Ghz.
The Pi4 with bluetooth devices, bluez 5.50-1.2~deb10u1+rpt1 and brcm80211 rpt5 is working now.

But I noticed that the Pi4 has a limited ability to connect.
On a 5Ghz ac WIFI it connects only with:
"n/ 40 MHz, WPA2, 1 x 1, 11k, 11v"

Jefraiboni, 2020-04-10:
Hi,

I am sorry my bad English, I have a pi 4, I am having an issue, I have a speaker jbl go, jbl charge 3 and handsfree Huawei Am61, My devices work when I power off Wifi on the pi 4, but I power on wifi my devices star connect and desconnect by ifselt that happens when I am not listening to music, My pi 4 is full upgrade, can you help me with a solution for this problem.

Regards, 

John

rtheil, 2020-04-13:
I'm having this same issue with a cheap bluetooth keyboard (https://www.amazon.com/gp/product/B00X5P8BJO) and with Apple bluetooth keyboard. Fresh install of Raspbian buster image on rpi4 (dated 2/13/20). Keyboard works perfectly. Updates are run and keyboard stops working. Connectivity is perfect. Keyboard always connects to Pi and shows as connected.

I do not have ability to switch between 2.4GHz and 5GHz on wifi as they are configured together on same SSID and many devices in the house still require 2.4GHz. However, I did disable wifi and the keyboards work perfectly. Problem is, next to my SSID it shows "5G" while connected.

pelwell, 2020-04-17:
Cypress have responded with an easy patch to the "NVRAM" configuration file. I've pushed the updated file to this repo and we'll get a Raspbian package out as soon as possible, but the easiest way to fix it for you self is to run this command:
```
$ echo btc_params50=0x972c | sudo tee -a /lib/firmware/brcm/brcmfmac43455-sdio.txt
```
You can confirm that the patch has been applied successfully with:
```
pi@raspberrypi:~$ md5sum /lib/firmware/brcm/brcmfmac43455-sdio.txt
0ed2738fb42c392c60e34dedb74d0510  /lib/firmware/brcm/brcmfmac43455-sdio.txt
```
Now reboot, and you should find it all works _much_ better.

XECDesign, 2020-04-17:
Should I push out a new package with that one line added, or is there more to it than that?

pelwell, 2020-04-17:
Yes please, there is nothing more to it than that.

macmpi, 2020-04-17:
I see current RPi-Distro firmware file is at 7.45.202 rather than latest 7.45.206

Is that supposed to work with latest [firmware files from Cypress](https://community.cypress.com/docs/DOC-19375):
File: firmware/brcmfmac43455-sdio.bin
Version: 7.45.206

pelwell, 2020-04-17:
As I mentioned in the message associated with [the commit](https://github.com/RPi-Distro/firmware-nonfree/commit/912b9ec80b063b3a30d7eb99cc6ab02b0c60c2d1), Cypress say this extra parameter is required for firmwares starting with 7.45.177. By using a default Cypress build you are missing out on some Pi-specific settings (and support from us), but adding this parameter might make a difference.

XECDesign, 2020-04-17:
Thanks Phil, it should be live within an hour.

pelwell, 2020-04-17:
Awesome.

pelwell, 2020-04-17:
firmware-brcm80211 1:20190114-1+rpt6 is now live. Thanks, @XECDesign.

djibux, 2020-04-17:
I can confirm it fixes the issue for me.

Thanks.

rtheil, 2020-04-18:
I can also confirm that this issue is fixed after running updates. Thank you!

akicker, 2020-04-18:
With the brcm80211 fw-update to +rpt6:
pi3 buster, latest blue* (+rpt1) --> OK :+1: 
pi4 buster, old blue* (5.50-1.2\~deb10u1) --> OK :+1: 
**pi4 buster, after unholding and updating blue* (5.50-1.2\~deb10u1+rpt1) --> Wlan again slowing down dramatically when BT is on :-1:  **

Jefraiboni, 2020-04-18:
Hi,
I'm sorry my bad English, me too can confirm that this issue is fixed after update to firmware-brcm80211 1:20190114-1+rpt6, thank you so much

pelwell, 2020-04-18:
@akicker There is going to be some contention in the 2.4G band - there are limits to what you can achieve with a shared antenna. Use a 5G WLAN for best performance. The difference between the deb10u1 and deb10u1+rpt1 is the code that gives bluetooth priority when streaming audio. If you aren't streaming audio then it shouldn't make a difference.

akicker, 2020-04-18:
@pelwell OK, thank you for this Info!
I am using BT and 2.4G-Wlan (my router doesnt support 5G), knowing that BT uses the same frequency range.
I dont use audio streaming, but my PI scans for BLE-devices, used as presence detection.
So I will stick with my well working solution - hiding blue*-updates ;-) 
PS: my live system runs on pi3b+ **stretch** - never made such trobles ;-)

macmpi, 2020-04-21:
@pelwell 
After careful comparison of new release with upstream NVRAM files (mostly an alphabetic reordering & different filenames), it turns-out that:
- `brcmfmac43455-sdio.txt` has **`btc_mode=1` parameter twice (typo)**, and differs from upstream `brcmfmac43455-sdio.raspberrypi,4-model-b.txt` by newly introduced `btc_params50=0x972c` (and `NVRAMRev` number)
(note: expectingly `upstream brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt` has same difference, plus `boardflags3`difference as actual boards are more instantiated upstream)

- `brcmfmac43430-sdio.txt` is a copycat of upstream `brcmfmac43430-sdio.raspberrypi,3-model-b.txt`. (upstream is astonishingly missing `brcmfmac43430-sdio.raspberrypi,model-zero-w.txt`, which would need a symlink or copy of 3-model-b)

Except for this new Pi4/Pi3b+ `btc_params50` (which I hope will show-up upstream soon?), I guess Pi Foundation prefers keeping by-feature ordering, rather than lining-up to upstream?

marquesn, 2020-05-01:
Hello everyone, I've also been having issues with bluetooth connectivity dropping when WiFi is enabled. I have Raspbian 9 stretch with Bluez 5.50 and hardware BCM2835, revision a020d3.

As @pelwell mentioned, this update should do the trick, but how exactly do I perform the update? I've already tried it with "sudo apt update" and "sudo apt upgrade", but the issues persist. Do I have to upgrade the distro to Buster? Or is there another way to apply the update?

Thanks!

pelwell, 2020-05-01:
What do the following two commands report?:
```
$ apt info firmware-brcm80211 | grep Version
$ strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version
```
If you've already updated and upgraded and these aren't showing:
```
Version: 1:20190114-1+rpt6
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv Version: 7.45.202 (r724630 CY) CRC: 4b9a9ceb Date: Mon 2020-03-02 23:32:43 PST Ucode Ver: 1043.2139 FWID 01-72f6ece2
```
then you'll have to either switch to Buster or (and this is much safer and easier) just overwrite the firmware directly:
```
$ wget https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm/brcmfmac43455-sdio.bin
$ wget https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm/brcmfmac43455-sdio.txt
$ wget https://github.com/RPi-Distro/firmware-nonfree/raw/master/brcm/brcmfmac43455-sdio.clm_blob
$ sudo cp brcmfmac43455-sdio.* /lib/firmware/brcm
```

marquesn, 2020-05-01:
@pelwell 

Many thanks for the prompt reply!

1. Running: 
`apt info firmware-brcm80211 | grep Version` 
results in the following: 
`WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
E: Invalid operation info
`

2. Running:
`strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version`
Results in:
`43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv Version: 7.45.154 (r684107 CY) CRC: b1f79383 Date: Tue 2018-02-27 03:18:17 PST Ucode Ver: 1043.2105 FWID 01-4fbe0b04`

So based on these results, should I just overwrite the firmware directly with the links that you suggested?

Thanks again!

pelwell, 2020-05-01:
> should I just overwrite the firmware directly with the links that you suggested?

I think so - my "apt fu" in strong enough to do it any other way, and it should be a safe operation.

marquesn, 2020-05-01:
@pelwell 

I've downloaded the files and copied them over to the /lib/firmware/brcm path. I'll also add that I did not replace or delete any other files that were already in /lib/firmware/brcm.

I've also rebooted the RPi3 b+ but the issue still persists, with bluetooth connections being disrupted constantly if WiFi is turned on (they work with WiFi off).

I ran the strings command again, and this time it returned the same result as yours:
`43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv Version: 7.45.202 (r724630 CY) CRC: 4b9a9ceb Date: Mon 2020-03-02 23:32:43 PST Ucode Ver: 1043.2139 FWID 01-72f6ece2`

Although the `apt info firmware-brcm80211 | grep Version` still returns the same "invalid operation info" error.

Should I attempt the buster upgrade instead? 

Thanks!

pelwell, 2020-05-01:
You can confirm the running version with:
```
$ dmesg | grep brcmfmac
[    3.922582] brcmfmac: F1 signature read @0x18000000=0x15264345
[    3.938684] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    3.939297] usbcore: registered new interface driver brcmfmac
[    4.259656] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.275308] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    6.135038] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
$ tail -1 /lib/firmware/brcm/brcmfmac43455-sdio.txt
btc_params50=0x972c
```
If you see the right versions there and it hasn't improved matters then I don't think updating to Buster will make much of a difference, but if you have a spare SD card then you could try a clean install and upgrade (allow the wizard to do the update) of the latest Buster image and quickly try WiFi and Bluetooth.

> just noticed I didn't mention that this is ocurring on a RPi 3 b+ in my initial post.

Yes you did - `revision a020d3` ;-)

marquesn, 2020-05-01:
@pelwell 

Thanks! I've ran those commands and they return the exact same results from the dmesg command, as well as the tail.

I guess I'll try out your suggestion with the buster upgrade and will post the results on here when I do. 

Thanks again!

marquesn, 2020-05-04:
Hello @pelwell 

I've done a fresh Buster install on the RPI 3 b+ and downloaded the new firmware files.

Bluetooth devices are still disconnecting when the WiFi is turned on. However this time the results of the dmesg grep command have detected some errors to do with scans? :
```
dmesg | grep brcmfmac
[    5.220747] brcmfmac: F1 signature read @0x18000000=0x15264345
[    5.234739] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.236739] usbcore: registered new interface driver brcmfmac
[    5.509316] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.544262] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[    8.120634] brcmfmac: power management disabled
[  315.591376] brcmfmac: brcmf_run_escan: error (-52)
[  315.591392] brcmfmac: brcmf_cfg80211_scan: scan error (-52)
[  442.856107] brcmfmac: power management disabled
```
The tail command:
```
tail -1 /lib/firmware/brcm/brcmfmac43455-sdio.txt
btc_params50=0x972c
```

Not quite sure what else I could do for now, so I'll continue searching online. Please let me know if you think of something else, and I will post on here if I find a solution.

Many Thanks!

akicker, 2020-05-06:
Hi, made some tests with my 3b+ Here the working configs:

RPI | OS | kernel | blue* | firmware-brcm80211
-- | -- | -- | -- | --
3Bplus | stretch | 4.19.66 | 5.43-2+rpt2+deb9u2 | +rpt4
3Bplus | buster | 4.19.97 | 5.50-1.2~deb10u1 | +rpt6

marquesn, 2020-05-06:
Hello @akicker 

My Bluez version is 5.50-1.2~deb10u1, but the firmware version is reported to be +rpt1. How do I go about upgrading it to rpt6?

Thanks!

akicker, 2020-05-06:
First of all hold upgrading bluetooth - see my second post from Mar 31. 
and full-upgrade should do the rest (upgrade it to rpt6) :-)

marquesn, 2020-05-06:
Hello @akicker 

I've placed the packages on hold as per your post on 31st March.

I followed this with a full upgrade. However, the command `apt show bluez-obexd` still shows that I'm still using rpt1:
```
apt show bluez-obexd
Package: bluez-obexd
Version: 5.50-1.2~deb10u1+rpt1
Priority: optional
Section: admin
Source: bluez
Maintainer: Debian Bluetooth Maintainers <team+pkg-bluetooth@tracker.debian.org>
Installed-Size: 473 kB
Depends: libc6 (>= 2.28), libdbus-1-3 (>= 1.9.14), libglib2.0-0 (>= 2.35.9), libical3 (>= 3.0.0)
Conflicts: obexd-client, obexd-server
Replaces: obexd-client, obexd-server
Homepage: http://www.bluez.org
Download-Size: 180 kB
APT-Sources: http://archive.raspberrypi.org/debian buster/main armhf Packages
```

akicker, 2020-05-06:
@marquesn dont worry 
try to show the installed version
dpkg --status bluez-obexd | grep '^Version:'
--> Version: 5.50-1.2~deb10u1

Swap-File, 2020-06-01:
I have a pair of Pi4s talking to each other via 5ghz wifi, one is running hostAPD.  I also have bluetooth traffic on the Pi running hostAPD.

Previously, I was running 1:20190114-1+rpt4 and 5.50-1+rpt1, and it worked well.

I recently ran apt update and noticed my wifi performance became very inconsistent.  My synced rate would go up and down wildly, and my connection would often drop entirely.

I tried rebuilding my cards from scratch with the new Raspberry Pi OS (32-bit) Lite image, and had the same problem.

I went back to my old image, and eventually found if I held back bluez, everything would keep working OK.

bluez 5.50-1+rpt1 with firmware-brcm80211 20190114-1+rpt6 is working well for me.

akicker, 2020-07-05:
After some tests on my pi4 kernel 4.19.118 using firmware-brcm80211 20190114-1+**rpt7** 
bluez 5.50-1.2\~deb10u1+**rpt1** still does **not** work :-1: 
My solution: sudo apt-mark hold bluez - update all others
using bluez 5.50-1.2\~deb10u1 works fine :+1:

DenisBY, 2020-08-02:
Just a side note. If I boot into Ubuntu 20.20 bluetooth works just fine. During scan I see all my devices and even devices of my neighbors. If I boot into 'buster' I can see only my MI Band 4 when I'm sitting close to Raspberry. 

In buster I have:
```
# strings /lib/firmware/brcm/brcmfmac43455-sdio.bin | grep Version
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-lpc-pwropt-43455_ftrs-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv Version: 7.45.202 (r724630 CY) CRC: 4b9a9ceb Date: Mon 2020-03-02 23:32:43 PST Ucode Ver: 1043.2139 FWID 01-72f6ece2
...
Unpacking firmware-brcm80211 (1:20190114-1+rpt5) over (1:20190114-1+rpt5)
...
```

akicker, 2020-08-05:
The latest bluez update rpt2 is better than rpt1 BUT
bluez 5.50-1.2\~deb10u1+rpt2 still slows down wlan :-1: 
using bluez 5.50-1.2\~deb10u1 still works fine :+1:

rob-meiner, 2020-08-06:
Dear colleagues,  Is there a definitive fix to the bluetooth disconnect problem?  I've read through all the forum entries I can find now and tried a lot of different things.  I have a raspberry pi 4b with 4G RAM, running Buster, headless.  I'm using a 2.4 Ghz WLAN.

Here's a[ link to the description of the problem I posted on the raspberrypi/firmware page](https://github.com/raspberrypi/firmware/issues/1447#issue-674248741)

I would really appreciate some insight into this issue.  This seems like a big deal, since headless users can't really get the value of the bt radio.

-Mark

rob-meiner, 2020-08-06:
Just to add a bit more information to this.  It appears that I am getting a firmware loading error. 

```
pi@antonio:~ $ dmesg | grep brcmfmac
[    4.785330] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.796418] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.796839] usbcore: registered new interface driver brcmfmac
[    4.818617] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt failed with error -2
[    5.046562] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.067193] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: May 20 2020 00:36:51 version manifest (de496ab CY) FWID 01-12aa6092
[    7.204682] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[    8.355542] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
pi@antonio:~ $ md5sum /lib/firmware/brcm/brcmfmac43455-sdio.bin
0a45a7a828e49e20fee2fe0438149523  /lib/firmware/brcm/brcmfmac43455-sdio.bin
```
I have tried updating the firmware per @pelwell in this post:  https://github.com/raspberrypi/firmware/issues/1403#issuecomment-638701200

Please let me know if you have any ideas how to resolve this.
Thanks, Mark

thamht4190, 2020-08-13:
For one who don't want to read all comments:
I got this problem on my Pi4B. I got the firmware +rpt7, so I suppose the fix (on +rpt6) is already there. If your firmware version is less than +rpt6, you should update it.

```
pi@raspberrypi:~ $ apt info firmware-brcm80211 | grep Version
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.
Version: 1:20190114-1+rpt7
```
I checked bluez version:

```
pi@raspberrypi:~ $ dpkg --status bluez \| grep '^Version:'
Version: 5.50-1.2~deb10u1+rpt1
```

From the comment on this ticket, I understand that **bluez version 5.50-1.2~deb10u1** can work, so I download it from [here](https://packages.debian.org/buster/armhf/bluez/download), then install it (bluez will be downgraded automatically):

```
pi@raspberrypi:~ $ wget http://ftp.hk.debian.org/debian/pool/main/b/bluez/bluez_5.50-1.2~deb10u1_armhf.deb
pi@raspberrypi:~ $ sudo apt install ./bluez_5.50-1.2~deb10u1_armhf.deb
```

And now, bluetooth and wifi can work together on my Pi.

stuckinger, 2020-10-04:
Thanks for this post! It kind of helped me, but only partly.
I want to share my experience because I got an improvement, someone else might be interested in.
Even though I end up using an extra usb wifi - read below why...

The Hardware:
```
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4`

Hardware        : BCM2835
Revision        : a02082
Model           : Raspberry Pi 3 Model B Rev 1.2
```

I do not have an exact setup from before when it was not working, but I was on stretch and on blue*-version 5.50 something I guess and kernel 4.19 something.
I had scratches in my audio in general, not only bluetooth.
Network via WIFI onboard and Bluetooth via onboard.
The main problem was, when trying to read/write Bluetooth LE Devices it mostly did not work:
```
Error: connect error: Function not implemented (38)
[E0:E5:CF:D6:AF:49][LE]> Bluetooth connection timed out
```

After reading this thread I updated to buster, so now my versions are:
```
~# uname -a
Linux  5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
```
```
~# dpkg -l |grep -i blue
ii  bluej                                 4.2.1                                     all          A simple but powerful Java IDE.
ii  blueman                               2.0.8-1+b2                                armhf        Graphical bluetooth manager
ii  bluetooth                             5.50-1.2~deb10u1+rpt2                     all          Bluetooth support (metapackage)
ii  bluez                                 5.50-1.2~deb10u1+rpt2                     armhf        Bluetooth tools and daemons
ii  bluez-alsa:armhf                      4.99-2+deb7u1                             armhf        Bluetooth ALSA support
ii  bluez-audio                           4.99-2+deb7u1                             all          Transitional package
ii  bluez-firmware                        1.2-4+rpt5                                all          Firmware for Bluetooth devices
ii  bluez-gstreamer                       4.99-2+deb7u1                             armhf        Bluetooth GStreamer support
ii  bluez-hcidump                         5.50-1.2~deb10u1+rpt2                     armhf        Analyses Bluetooth HCI packets
ii  bluez-obexd                           5.50-1.2~deb10u1+rpt2                     armhf        bluez obex daemon
ii  bluez-tools                           2.0~20170911.0.7cb788c-2                  armhf        Set of tools to manage Bluetooth devices for linux
ii  bluez-utils                           4.99-2+deb7u1                             all          Transitional package
ii  libbluetooth3:armhf                   5.50-1.2~deb10u1+rpt2                     armhf        Library to use the BlueZ Linux Bluetooth stack
ii  pi-bluetooth                          0.1.15                                    all          Raspberry Pi 3 bluetooth
ii  pulseaudio-module-bluetooth           12.2-4+deb10u1                            armhf        Bluetooth module for PulseAudio sound server

```
```
~# dpkg -l |grep -i firmware
ii  bluez-firmware                        1.2-4+rpt5                                all          Firmware for Bluetooth devices
ii  firmware-atheros                      1:20190114-1+rpt8                         all          Binary firmware for Atheros wireless cards
ii  firmware-brcm80211                    1:20190114-1+rpt8                         all          Binary firmware for Broadcom/Cypress 802.11 wireless cards
ii  firmware-libertas                     1:20190114-1+rpt8                         all          Binary firmware for Marvell wireless cards
ii  firmware-misc-nonfree                 1:20190114-1+rpt8                         all          Binary firmware for various drivers in the Linux kernel
ii  firmware-ralink                       1:20190114-1+rpt8                         all          Binary firmware for Ralink wireless cards (dummmy package)
ii  firmware-realtek                      1:20190114-1+rpt8                         all          Binary firmware for Realtek wired/wifi/BT adapters
ii  rpi-update                            20200409                                  all          Raspberry Pi firmware updating tool
```

In first place I was really happy because even though my pi was still connected via internal WIFI I could control my bluetooth devices!
I even found out that the sound was way better than before - I haven't tested bluetooth yet, only audio plug.
But then I realized that there are still some conflicts because while streaming audio - I have an mpd running that is streaming from network - every time I connect to a bluetooth-device the music stopped. I guess it comes from buffer underrun, maybe the wifi connection drops for some seconds. Something I did not realize while being on ssh.

So while not using it as a radio station I guess this setup would work fine, but constant WIFI network AND Bluetooth still conflicts.

My workaround: Using USB WIFI.
Sound = OK
Bluetooth = OK
WIFI = OK

Still sad since the hardware is already onboard and should be working...

sebastianeferrer, 2021-01-29:
Hi everyone! great news! we finally have an official bugfix for CYW43455.
```
$ apt changelog bluez-firmware | head

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Get:1 store: bluez-firmware 1.2-4+rpt8 Changelog
bluez-firmware (1.2-4+rpt8) buster; urgency=medium

  [ Phil Elwell ]
  * Second Spectra fix for CYW43455 (CVE-2020-10370)
    - Original firmware name: BCM4345C0_003.001.025.0174.0342.hcd

 -- Serge Schneider <serge@raspberrypi.com>  Thu, 28 Jan 2021 14:57:44 +0000
```

Just run:

`$ sudo apt update && sudo apt upgrade bluez-firmware`

then it should work again. 

May be a deeper review on different implementations should be done, but first signs from me are good. My problem of interference between bt and wifi working on 2.4GHz was fixed

akicker, 2021-01-30:
Great, that's working, did remove my hold on bluez (sudo apt-mark unhold bluez) updated all and everything is OK again :-)

pelwell, 2021-01-30:
I'm going to close this issue then.

closed 2021-01-30 by pelwell

rabe-, 2021-02-16:
I'm still having that problem on my Pi4 with the K380

pelwell, 2021-02-16:
That's strange - I'm using one now and it works fine. Which distribution are you using?

lesinigo, 2021-04-07:
I'm having the same problem:
* Raspberry Pi Zero W rev1.1
* fresh Raspbian buster + `apt-get update; apt-get dist-upgrade`
* kernel `5.10.17+ #1403`, bluez `5.50-1.2~deb10u1+rpt2`, bluez-firmware `1.2-4+rpt8`

After weeks without any problems while using wifi only, I turned on bluetooth and suddenly got connectivity issues over WiFi (2.4GHz). Disabling bluetooth gave me a perfectly working wifi again.

These are from my dmesg:
```
[   17.552652] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   17.941396] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   18.363915] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   18.376905] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   18.377393] usbcore: registered new interface driver brcmfmac
[   18.412091] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
[   18.652728] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   18.652945] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   18.654148] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   27.978848] uart-pl011 20201000.serial: no DMA platform data
[   29.175921] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   31.900322] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   32.531216] Bluetooth: Core ver 2.22
[   32.531389] NET: Registered protocol family 31
[   32.531405] Bluetooth: HCI device and connection manager initialized
[   32.531440] Bluetooth: HCI socket layer initialized
[   32.531462] Bluetooth: L2CAP socket layer initialized
[   32.531510] Bluetooth: SCO socket layer initialized
[   32.554015] Bluetooth: HCI UART driver ver 2.3
[   32.554045] Bluetooth: HCI UART protocol H4 registered
[   32.554212] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   32.554500] Bluetooth: HCI UART protocol Broadcom registered
[   33.914212] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   33.914238] Bluetooth: BNEP filters: protocol multicast
[   33.914272] Bluetooth: BNEP socket layer initialized
```