RPi-Distro/firmware-nonfree #9: firmware-brcm80211 1:20190114-1+rpt5 causes failure to connect to some WPA2 APs

JoeyDodson, 2020-04-03:
Running Buster Lite (2020-02-13-raspbian-buster-lite) on Rpi4, I was able to connect to an AP running on my laptop using (hostapd), but after upgrading from firmware-brcm80211 1:20190114-1+rpt4 to firmware-brcm80211 1:20190114-1+rpt5, the Rpi4 fails to connect to the AP. It can still connect to my home wifi, however. 

I detailed the problem, included logs and configuration files [on Stack Exchange](https://raspberrypi.stackexchange.com/questions/110786/rpi4-with-buster-wpa-supplicant-fails-to-connect-to-hostapd-network-hosted-on-ub/111001#111001).

pelwell, 2020-04-04:
It would have been helpful to summarise the results is that thread, which I take to be that you are running hostapd on your laptop, and with the latest firmware the AP is rejecting the connection with the following error message:
```
WPA: invalid MIC in msg 2/4 of 4-Way Handshake
```
Is that correct, and is there anything
*

JoeyDodson, 2020-04-04:
> It would have been helpful to summarise the results in that thread, which I take to be that you are running hostapd on your laptop, and with the latest firmware the AP is rejecting the connection with the following error message:
> 
> ```
> WPA: invalid MIC in msg 2/4 of 4-Way Handshake
> ```
> 
> Is that correct, and is there anything to add to that?

Yes, that is correct. I have edited the post to include the portions of logs showing the failure on each side of the connection.

pelwell, 2020-04-07:
I'm trying to reproduce this now, using a Pi 3 as an AP, and its working with the latest 43455 firmware.

Can you post a copy of the hostapd.conf (minus the passphrase) from the laptop so I can look for differences?

JoeyDodson, 2020-04-10:
From my laptop:
/etc/hostapd/hostapd.conf

`##### hostapd configuration file ##############################################
# Empty lines and lines starting with # are ignored

# AP netdevice name (without 'ap' postfix, i.e., wlan0 uses wlan0ap for
# management frames with the Host AP driver); wlan0 with many nl80211 drivers
# Note: This attribute can be overridden by the values supplied with the '-i'
# command line parameter.
interface=wlx2824ff1a0a09
# In case of atheros and nl80211 driver interfaces, an additional
# configuration parameter, bridge, may be used to notify hostapd if the
# interface is included in a bridge. This parameter is not used with Host AP
# driver. If the bridge parameter is not set, the drivers will automatically
# figure out the bridge interface (assuming sysfs is enabled and mounted to
# /sys) and this parameter may not be needed.
#
# For nl80211, this parameter can be used to request the AP interface to be
# added to the bridge automatically (brctl may refuse to do this before hostapd
# has been started to change the interface mode). If needed, the bridge
# interface is also created.
#bridge=br0

# Driver interface type (hostap/wired/none/nl80211/bsd);
# default: hostap). nl80211 is used with all Linux mac80211 drivers.
# Use driver=none if building hostapd as a standalone RADIUS server that does
# not control any wireless/wired driver.
driver=nl80211

# Driver interface parameters (mainly for development testing use)
# driver_params=<params>

# hostapd event logger configuration
#
# Two output method: syslog and stdout (only usable if not forking to
# background).
#
# Module bitfield (ORed bitfield of modules that will be logged; -1 = all
# modules):
# bit 0 (1) = IEEE 802.11
# bit 1 (2) = IEEE 802.1X
# bit 2 (4) = RADIUS
# bit 3 (8) = WPA
# bit 4 (16) = driver interface
# bit 5 (32) = IAPP
# bit 6 (64) = MLME
#
# Levels (minimum value for logged events):
#  0 = verbose debugging
#  1 = debugging
#  2 = informational messages
#  3 = notification
#  4 = warning
#
logger_syslog=-1
logger_syslog_level=0
logger_stdout=-1
logger_stdout_level=0

# Interface for separate control program. If this is specified, hostapd
# will create this directory and a UNIX domain socket for listening to requests
# from external programs (CLI/GUI, etc.) for status information and
# configuration. The socket file will be named based on the interface name, so
# multiple hostapd processes/interfaces can be run at the same time if more
# than one interface is used.
# /var/run/hostapd is the recommended directory for sockets and by default,
# hostapd_cli will use it when trying to connect with hostapd.
ctrl_interface=/var/run/hostapd

[... 1923 lines cut ...]
# - results in a valid MASK that covers it and the dev_addr
# - is not the same as the MAC address of the radio
# - is not the same as any other explicitly specified BSSID
#
# Alternatively, the 'use_driver_iface_addr' parameter can be used to request
# hostapd to use the driver auto-generated interface address (e.g., to use the
# exact MAC addresses allocated to the device).
#
# Not all drivers support multiple BSSes. The exact mechanism for determining
# the driver capabilities is driver specific. With the current (i.e., a recent
# kernel) drivers using nl80211, this information can be checked with "iw list"
# (search for "valid interface combinations").
#
# Please note that hostapd uses some of the values configured for the first BSS
# as the defaults for the following BSSes. However, it is recommended that all
# BSSes include explicit configuration of all relevant configuration items.
#
#bss=wlan0_0
#ssid=test2
# most of the above items can be used here (apart from radio interface specific
# items, like channel)

#bss=wlan0_1
#bssid=00:13:10:95:fe:0b
# ...`

nerdlinger, 2020-06-17:
I can confirm this problem occurred on a Pi 3A+.

I ended up reverting to the latest stable Broadcom firmware by running:

`sudo apt upgrade firmware-brcm80211/stable`

I have also prevented upgrading to the testing branch by placing the following in /etc/apt/preferences:

```
Package: firmware-brcm80211
Pin: release a=stable
Pin-Priority: 700
```

nerdlinger, 2020-07-02:
Issue persists with firmware-brcm80211 1:20190114-1+rpt7, publicly released yesterday.