RPi-Distro/firmware-nonfree #16: RasPi Zero W: WiFi NOT working: ieee80211 phy0: brcmf_c_process_clm_blob: clmload (4733 byte file) failed (-52)

colemickens, 2021-09-29:
**Raspberry Pi Zero W**
**+ Latest RPi Foundation Kernel**
**+ Latest wireless firmware from this repo**

**= Not working**

I think this is the fatally bad part:
`ieee80211 phy0: brcmf_c_process_clm_blob: clmload (4733 byte file) failed (-52)`

I grabbed a usb-ethernet adapter, built a NixOS image with tailscale, and got access to it to grab dmesg logs:

```
[   38.672230] 8021q: 802.1Q VLAN Support v1.8
[   40.721334] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   41.013255] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   42.963928] usb_phy_generic phy: supply vcc not found, using dummy regulator
[   45.725657] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[   45.961132] mc: Linux media interface: v0.10
[   46.158245] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   46.647926] [vc_sm_connected_init]: start
[   46.959650] [vc_sm_connected_init]: installed successfully
[   47.195627] Error: Driver 'sdhost-bcm2835' is already registered, aborting...
[   49.302148] videodev: Linux video capture interface: v2.00
[   50.449146] tun: Universal TUN/TAP device driver, 1.6
[   50.939855] asix 1-1:1.0 eth0: register 'asix' at usb-20980000.usb-1, ASIX AX88772B USB 2.0 Ethernet, 00:0e:c6:85:11:33
[   50.985012] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   51.131625] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   51.157174] usbcore: registered new interface driver asix
[   51.538717] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   51.593716] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[   51.849464] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   52.239245] bcm2835-isp bcm2835-isp: Device node output[0] registered as /dev/video13
[   52.280344] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   52.323389] bcm2835-isp bcm2835-isp: Device node capture[0] registered as /dev/video14
[   52.365572] bcm2835-isp bcm2835-isp: Device node capture[1] registered as /dev/video15
[   52.366472] bcm2835-isp bcm2835-isp: Device node stats[2] registered as /dev/video16
[   52.366722] bcm2835-isp bcm2835-isp: Register output node 0 with media controller
[   52.366931] bcm2835-isp bcm2835-isp: Register capture node 1 with media controller
[   52.367147] bcm2835-isp bcm2835-isp: Register capture node 2 with media controller
[   52.367378] bcm2835-isp bcm2835-isp: Register capture node 3 with media controller
[   52.414858] bcm2835-isp bcm2835-isp: Loaded V4L2 bcm2835-isp
[   52.499003] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   52.510004] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   52.510311] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   52.595979] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   52.596296] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   52.750565] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   52.750900] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   52.906778] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   52.936167] usbcore: registered new interface driver brcmfmac
[   53.413369] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   53.413944] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   53.422678] ieee80211 phy0: brcmf_c_process_clm_blob: clmload (4733 byte file) failed (-52)
[   53.428560] ieee80211 phy0: brcmf_c_preinit_dcmds: download CLM blob file failed, -5
[   53.428872] ieee80211 phy0: brcmf_bus_started: failed: -5
[   53.429095] ieee80211 phy0: brcmf_attach: dongle is not responding: err=-5
[   53.440324] brcmfmac: brcmf_sdio_firmware_callback: brcmf_attach failed
[   65.113095] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   65.123239] asix 1-1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   65.169527] asix 1-1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
rpizero2% uname -a
Linux rpizero2 5.10.52 #1-NixOS Tue Jan 1 00:00:00 UTC 1980 armv6l GNU/Linux
```

Versions:
* **linux**: [raspberrypi/linux@1.20210805](https://github.com/raspberrypi/linux/tree/1.20210805)
* **raspberrypi-wireless-firmware**: [RPiDistro/firmware-nonfree@00de3194a96397c913786945ac0af1fd6fbec45b](https://github.com/RPi-Distro/firmware-nonfree/tree/00de3194a96397c913786945ac0af1fd6fbec45b)

Extra Detail:
   * from our nixpkgs package, [this is what we copy](https://github.com/NixOS/nixpkgs/blob/42d7abfe9cb723757a758e55970d75668a161ef5/pkgs/os-specific/linux/firmware/raspberrypi-wireless/default.nix):
      ```
      for filename in firmware-nonfree/brcm/brcmfmac434??-sdio.*; do
        cp "$filename" "$out/lib/firmware/brcm"
      done
      ```

I tested briefly, using mainline linux kernel was broken in the same way or very similar.

colemickens, 2021-09-29:
I suspect I'm not the only one to have noticed this:
* this user says they've also experienced regressions around firmware/rpi0w, has multiple +1s: https://github.com/raspberrypi/linux/issues/4145#issuecomment-778832112
* ALARM is patching for this: https://github.com/archlinuxarm/PKGBUILDs/blob/3a28adcf6b76806891774a1d767824a649da51c6/alarm/firmware-raspberrypi/PKGBUILD#L22-L23

Of course, I have no idea where ALARM is getting that file from, or why they're also seemingly unable to use the one this repo provides.

pelwell, 2021-09-29:
What do you get from `ls -l /lib/firmware/brcm/brcmfmac43430*`? The 43430 firmware has a built in CLM database so doesn't need an external clm_blob file - notice that this repo doesn't contain one.

pelwell, 2021-09-29:
I note that the Linux firmware repo includes a `cypress/cyfmac43430-sdio.clm_blob` (md5 9ff46519b8b8c2cab323322c9d983873). If you find a `brcmfmac43430-sdio.clm_blob`, what is its `md5sum`?

pelwell, 2021-09-29:
A fresh install of the latest RPiOS image gives working WLAN on the Zero W. Allowing it to self-update (not a quick process) fails to break it. For comparison, after a reboot the kernel log contains:
```
$ dmesg | grep brcmfmac
[   23.762658] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[   23.992846] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   24.004350] usbcore: registered new interface driver brcmfmac
[   24.369277] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   24.369616] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   24.369808] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   24.371193] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 22 2019 01:59:28 version 7.45.98.94 (r723000 CY) FWID 01-3b33decd
[   48.592692] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

colemickens, 2021-09-29:
```
rpizero2% ls -l /run/current-system/firmware/brcm/brcmfmac43430*
lrwxrwxrwx 1 root root 120 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430a0-sdio.bin -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430a0-sdio.bin
lrwxrwxrwx 1 root root 131 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430a0-sdio.ilife-S806.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430a0-sdio.ilife-S806.txt
lrwxrwxrwx 1 root root 139 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430a0-sdio.jumper-ezpad-mini3.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430a0-sdio.jumper-ezpad-mini3.txt
lrwxrwxrwx 1 root root 134 Jan  1  1970 '/run/current-system/firmware/brcm/brcmfmac43430a0-sdio.ONDA-V80 PLUS.txt' -> '/nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430a0-sdio.ONDA-V80 PLUS.txt'
lrwxrwxrwx 1 root root 163 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.AP6212.txt -> /nix/store/xarcz7nl4l4wdx2s81zr0rr2sqgqlaq0-raspberrypi-wireless-firmware-armv6l-unknown-linux-gnueabihf-2021-06-28/lib/firmware/brcm/brcmfmac43430-sdio.AP6212.txt
lrwxrwxrwx 1 root root 156 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.bin -> /nix/store/xarcz7nl4l4wdx2s81zr0rr2sqgqlaq0-raspberrypi-wireless-firmware-armv6l-unknown-linux-gnueabihf-2021-06-28/lib/firmware/brcm/brcmfmac43430-sdio.bin
lrwxrwxrwx 1 root root 123 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.clm_blob -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.clm_blob
lrwxrwxrwx 1 root root 174 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.Hampoo-D2D3_Vi8A1.txt -> /nix/store/xarcz7nl4l4wdx2s81zr0rr2sqgqlaq0-raspberrypi-wireless-firmware-armv6l-unknown-linux-gnueabihf-2021-06-28/lib/firmware/brcm/brcmfmac43430-sdio.Hampoo-D2D3_Vi8A1.txt
lrwxrwxrwx 1 root root 163 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.MUR1DX.txt -> /nix/store/xarcz7nl4l4wdx2s81zr0rr2sqgqlaq0-raspberrypi-wireless-firmware-armv6l-unknown-linux-gnueabihf-2021-06-28/lib/firmware/brcm/brcmfmac43430-sdio.MUR1DX.txt
lrwxrwxrwx 1 root root 140 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt
lrwxrwxrwx 1 root root 143 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt
lrwxrwxrwx 1 root root 172 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.raspberrypi-rpi.txt -> /nix/store/xarcz7nl4l4wdx2s81zr0rr2sqgqlaq0-raspberrypi-wireless-firmware-armv6l-unknown-linux-gnueabihf-2021-06-28/lib/firmware/brcm/brcmfmac43430-sdio.raspberrypi-rpi.txt
lrwxrwxrwx 1 root root 139 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-plus.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-plus.txt
lrwxrwxrwx 1 root root 140 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-ultra.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-ultra.txt
lrwxrwxrwx 1 root root 139 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-zero.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m2-zero.txt
lrwxrwxrwx 1 root root 134 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m3.txt -> /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.sinovoip,bpi-m3.txt
lrwxrwxrwx 1 root root 156 Jan  1  1970  /run/current-system/firmware/brcm/brcmfmac43430-sdio.txt -> /nix/store/xarcz7nl4l4wdx2s81zr0rr2sqgqlaq0-raspberrypi-wireless-firmware-armv6l-unknown-linux-gnueabihf-2021-06-28/lib/firmware/brcm/brcmfmac43430-sdio.txt
```

Indeed, the `clm_blob` is indeed coming from `firmware-linux-nonfree` rather than the `raspberrypi-wireless-firmware` package:

`/nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/brcm/brcmfmac43430-sdio.clm_blob`

And finally:
```
rpizero2% sha256sum /run/current-system/firmware/brcm/brcmfmac43430-sdio.clm_blob
3376b9c9b32d16bf762e21c7fafb665365070ae240d092498d0d1987c22022aa  /run/current-system/firmware/brcm/brcmfmac43430-sdio.clm_blob

rpizero2% sha256sum $(readlink -f /run/current-system/firmware/brcm/brcmfmac43430-sdio.clm_blob)
3376b9c9b32d16bf762e21c7fafb665365070ae240d092498d0d1987c22022aa  /nix/store/1qb4kf2jkkdw7mqzahfgnk4zmhxbvgv0-firmware-linux-nonfree-2021-08-18/lib/firmware/cypress/cyfmac43430-sdio.clm_blob
```

colemickens, 2021-09-29:
@pelwell I do understand that RPiOS is working though, I can eventually try to provision, update, and then image the SD card and try to figure out what the diffs are in terms of installed firmware files.

colemickens, 2021-09-29:
I'm guessing, though, that RPiOS is using old firmware packages...

For some extra info:
```
rpizero2% strings cyfmac43430-sdio.bin | grep 202
^[[AMar 30 2021
2021-03-30 01:02:19
```

colemickens, 2021-09-29:
I created a `firmware-linux-nonfree` package that drops out:
* `brcmfmac43430-sdio.clm_blob`
* `cypress43430-sdio.clm_blob`

Pushed the config over, rebooted, and now wifi works.

Where should this be fixed?

pelwell, 2021-09-29:
In whichever repo your OS is getting them from - I don't think it's our problem.

colemickens, 2021-09-29:
Uhm, that's upstream `linux-firmware`

colemickens, 2021-09-29:
I guess maybe that's my answer though? I guess I just assumed it would've been someone from... the Raspberry PI Foundation that would've pushed these blobs into linux-firmware in the first place, but maybe not? I just don't know who owns which pieces.

pelwell, 2021-09-29:
Before you report a problem with the upstream firmware repo, just be certain that your OS hasn't ended up with an incompatible combination of upstream and ours, which would make it an OS packaging problem.

colemickens, 2021-09-29:
@pelwell aha, I think that's kind of the question I'm dancing around, as an OS maintainer, how do I determine the expected compatibility matrix?

pelwell, 2021-09-30:
Just pick one source - upstream or downstream. There will be differences, but each _should_ work.