RPi-Distro/firmware-nonfree #17: brcmfmac43455-sdio.clm_blob @d27b9f1 (2.6kB) disables all DFS channels

kmarty, 2021-10-01:
Blob **_brcmfmac43455-sdio.clm_blob_** from commit https://github.com/RPi-Distro/firmware-nonfree/commit/d27b9f1ce094fcb49e30e0f9fcdc7a4459b9c211 disables all DFS channels, making 5GHz WiFi basically unusable. At least for country=CZ:
blob from https://github.com/RPi-Distro/firmware-nonfree/commit/1f85a05b7ef063291e7a3dd1d735df4cbc441a74  (i.e. working):
```
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (20.0 dBm)
			* 5190 MHz [38] (disabled)
			* 5200 MHz [40] (20.0 dBm)
			* 5210 MHz [42] (disabled)
			* 5220 MHz [44] (20.0 dBm)
			* 5230 MHz [46] (disabled)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (20.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (20.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (20.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (20.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (20.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (20.0 dBm) (no IR, radar detection)
			* 5620 MHz [124] (20.0 dBm) (no IR, radar detection)
			* 5640 MHz [128] (20.0 dBm) (no IR, radar detection)
			* 5660 MHz [132] (20.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (20.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (20.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (disabled)
			* 5765 MHz [153] (disabled)
			* 5785 MHz [157] (disabled)
			* 5805 MHz [161] (disabled)
			* 5825 MHz [165] (disabled)
```

blob from https://github.com/RPi-Distro/firmware-nonfree/commit/d27b9f1ce094fcb49e30e0f9fcdc7a4459b9c211 (i.e. faulty):
```
		Frequencies:
			* 5170 MHz [34] (disabled)
			* 5180 MHz [36] (20.0 dBm)
			* 5190 MHz [38] (disabled)
			* 5200 MHz [40] (20.0 dBm)
			* 5210 MHz [42] (disabled)
			* 5220 MHz [44] (20.0 dBm)
			* 5230 MHz [46] (disabled)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (disabled)
			* 5280 MHz [56] (disabled)
			* 5300 MHz [60] (disabled)
			* 5320 MHz [64] (disabled)
			* 5500 MHz [100] (disabled)
			* 5520 MHz [104] (disabled)
			* 5540 MHz [108] (disabled)
			* 5560 MHz [112] (disabled)
			* 5580 MHz [116] (disabled)
			* 5600 MHz [120] (disabled)
			* 5620 MHz [124] (disabled)
			* 5640 MHz [128] (disabled)
			* 5660 MHz [132] (disabled)
			* 5680 MHz [136] (disabled)
			* 5700 MHz [140] (disabled)
			* 5720 MHz [144] (disabled)
			* 5745 MHz [149] (20.0 dBm)
			* 5765 MHz [153] (20.0 dBm)
			* 5785 MHz [157] (20.0 dBm)
			* 5805 MHz [161] (20.0 dBm)
			* 5825 MHz [165] (20.0 dBm)
```

pelwell, 2021-10-02:
Thanks for the feedback - it will be passed on to Cypress. Attempts were made to verify that no legitimate channels were lost, but clearly something has gone wrong.

antoneliasson, 2023-08-02:
I have the opposite problem with the latest clm_blob on bullseye (which was added in https://github.com/RPi-Distro/firmware-nonfree/commit/dc406650e840705957f8403efeacf71d2d7543b3 ). At least channel 149 which my WiFi uses does not work with this clm_blob. However, with https://github.com/RPi-Distro/firmware-nonfree/commit/d27b9f1ce094fcb49e30e0f9fcdc7a4459b9c211 , the previous "faulty" clm_blob, channel 149 works.

This appears to be the case also for kmarty judging from the lines marked disabled above. For some reason `iw list` doesn't show any channels as disabled for me though. The ones that don't work simply don't show up in scans.

iw list output is identical with both clm_blobs:
```
Wiphy phy0
...
	Band 2:
...
		Frequencies:
			* 5170 MHz [34] (20.0 dBm)
			* 5180 MHz [36] (20.0 dBm)
			* 5190 MHz [38] (20.0 dBm)
			* 5200 MHz [40] (20.0 dBm)
			* 5210 MHz [42] (20.0 dBm)
			* 5220 MHz [44] (20.0 dBm)
			* 5230 MHz [46] (20.0 dBm)
			* 5240 MHz [48] (20.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (radar detection)
			* 5280 MHz [56] (20.0 dBm) (radar detection)
			* 5300 MHz [60] (20.0 dBm) (radar detection)
			* 5320 MHz [64] (20.0 dBm) (radar detection)
			* 5500 MHz [100] (20.0 dBm) (radar detection)
			* 5520 MHz [104] (20.0 dBm) (radar detection)
			* 5540 MHz [108] (20.0 dBm) (radar detection)
			* 5560 MHz [112] (20.0 dBm) (radar detection)
			* 5580 MHz [116] (20.0 dBm) (radar detection)
			* 5600 MHz [120] (20.0 dBm) (radar detection)
			* 5620 MHz [124] (20.0 dBm) (radar detection)
			* 5640 MHz [128] (20.0 dBm) (radar detection)
			* 5660 MHz [132] (20.0 dBm) (radar detection)
			* 5680 MHz [136] (20.0 dBm) (radar detection)
			* 5700 MHz [140] (20.0 dBm) (radar detection)
			* 5720 MHz [144] (13.0 dBm) (radar detection)
			* 5745 MHz [149] (13.0 dBm)
			* 5765 MHz [153] (13.0 dBm)
			* 5785 MHz [157] (13.0 dBm)
			* 5805 MHz [161] (13.0 dBm)
			* 5825 MHz [165] (13.0 dBm)
...
```

My regulatory domain is SE (Sweden). As far as I can tell Linux has the right regulatory domain set:
```
$ iw reg get
global
country SE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

phy#0
country 99: DFS-UNSET
	(2402 - 2482 @ 40), (6, 20), (N/A)
	(2474 - 2494 @ 20), (6, 20), (N/A)
	(5140 - 5360 @ 160), (6, 20), (N/A)
	(5460 - 5860 @ 160), (6, 20), (N/A)
```

pelwell, 2023-08-05:
I can confirm your findings - channels above 140 seem to be inaccessible. The same applies to GB, BR, and probably many others. The only country I've found so far that can see 144, 149, 153 and 157 is Russia (RU), but that is at the expense of channels 100-128 which disappear. I also found that connecting to my AP forced the country to GB, so you may find that forcing RU isn't even a workaround.

I find it interesting that both the 43455 and the 43456 (the sister chip from Synaptics) behave similarly (although even on RU, 43456 can only get as high as 144), making me wonder if there is something more than a simple oversight at work here. I'll ask Infineon.

pelwell, 2023-08-07:
I find that setting my AP to US allows it to use channel 149, then after setting the Pi also to US it will connect. This suggests that it isn't a fundamental problem with the firmware, but that the clm_blob file (which tells the firmware which channels are available) may not be enabling all the channels it should.