RPi-Distro/firmware-nonfree #20: Please restore master branch

rzr, 2021-11-18:
Hi,

It looks like master branch have been removed recently ? do you confirm ? Is it intentional or a mistake ?

I am asking because other projects are referring to it:
https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb

Should reference be updated to align to debian named branches  (IMHO tags would be good to use too).  

Any recommendations welcome.

Thx

XECDesign, 2021-11-18:
The previous `master` branch was renamed to `buster`, but I'd recommend switching over to `bullseye`, although it's structured a bit differently.

korya, 2021-12-06:
This change caused our build to start fail as well. We use https://git.yoctoproject.org/meta-raspberrypi. I assume we will have to upgrade it now.

Is there an easy way to change `SRC_URI` in https://git.yoctoproject.org/meta-raspberrypi/tree/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro_git.bb?h=dunfell#n36 via `.bbapend`? I would like to avoid upgrading the whole `meta-raspberrypi`.

XECDesign, 2021-12-07:
The `master` branch was quite different to the currently maintained `bullseye` branch, so a change to would be required anyway.

Given that this repo was intended for the Debian package, it's a bit hard to use it as a stable source for third parties. We'll have issues again when bookworm comes out.

@pelwell Would it make sense to have a separate repo that's intended for distribution rather than packaging? Something where branches and paths won't change?

pelwell, 2021-12-07:
I wouldn't have a problem with putting both in the same repo, whether by replication or symbolic links (if that could work), i.e. retain the buster structure alongside the packaging info, but splitting the two is also OK.

Actually, creating a firmware repo under the `raspberrypi` Github organisation would solve another problem at the same time. We should discuss this offline.

colemickens, 2022-05-02:
I *still* get confused every time I try to do updates for NixOS. It would be great if there were clear answers for maintainers. We appear to have maybe just silently broken wifi for folks....

XECDesign, 2022-05-03:
> It would be great if there were clear answers for maintainers.

What's the question?