RPi-Distro/firmware-nonfree #23: Raspberry Pi Zero 2 W: BCM43430/2 firmware does not support 4-way handshake offloading

stapelberg, 2022-03-24:
It seems like there are (at least) two different Raspberry Pi Zero 2 W revisions on the market, and they differ in terms of which WiFi chip is used.

The BCM43430/1 works as expected, and seems to match the chip used on the Raspberry Pi 3 B.

The BCM43430/2 however requires different firmware files (brcmfmac43436). This firmware unfortunately lacks support for 4-way handshake offloading, as can be verified by running `iw phy`, which prints:
```
	Supported extended features:
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
```

…whereas on the BCM43430/1, Raspberry Pi 3B, 3B+ or 4B, it prints:
```
	Supported extended features:
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
		* [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
		* [ SAE_OFFLOAD ]: SAE offload support
```

The 4-way handshake offloading is not just a performance optimization, but rather the feature that made using encrypted WiFi possible at all on platforms like https://gokrazy.org/ (not based on Raspberry Pi OS, entirely implemented in Go aside from the Linux kernel). With the offloading, connecting to WiFi is as simple as one Linux nl80211 API call, whereas without the offloading, a lot of work is required in userspace (`wpa_supplicant`).

Is the lack of 4-way handshake offloading an oversight in the current firmware blobs for the BCM43430/2?

Is there any chance we could get a fixed firmware version?

Thanks in advance

(PS: we figured this out in https://github.com/gokrazy/gokrazy/issues/96, but I’m filing a separate issue as the history is getting lengthy…)

pelwell, 2022-03-24:
Thanks for collating the significant details in one place. A ticket has been opened with Synaptics.

stapelberg, 2022-03-24:
Thanks, Phil! I appreciate it.

joneskoo, 2022-03-25:
@pelwell In the meanwhile, do you have information how we can identify Zero 2 W with each chip so we can document compatibility?

pelwell, 2022-03-25:
The chips are featureless and under the metal cap of the module so there is no visible difference, and the firmware doesn't know - the board revisions are identical; only the kernel can tell after probing the SDIO bus when booting. The best way to tell which version you have to is to boot it up and type:
```
$ dmesg | grep "Firmware: BCM43430"
```
If the output includes `BCM43430/2` (corresponding to the `brcmfmac43436-sdio` firmware) the part _is_ affected by this issue.
If the output includes `BCM43430/1` (corresponding to the `brcmfmac43436s-sdio` firmware) then the part _is not_ affected.

pelwell, 2022-03-30:
Synaptics have given us a trial firmware. It appears to make the right noise:
```
        Supported extended features:
                * [ 4WAY_HANDSHAKE_STA_PSK ]: 4-way handshake with PSK in station mode
                * [ 4WAY_HANDSHAKE_STA_1X ]: 4-way handshake with 802.1X in station mode
                * [ SAE_OFFLOAD ]: SAE offload support
```
The version message at boot is:
```
[    8.450230] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    8.508391] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Mar 29 2022 19:22:43 version 9.88.4.77 (g58bc5cc) FWID 01-623162e
```
You will have to rename the firmware when you install it, but I don't think you'll struggle with that...

https://drive.google.com/file/d/1_q2aXxBV5PTUSVAE2crt4uqfjbN9ebN3/view?usp=sharing

stapelberg, 2022-03-30:
Thanks for relaying the trial firmware!

I have just verified it on my Raspberry Pi Zero 2 W, and it indeed works! I can now connect to encrypted WiFi, too.

This is great news. Thanks so much for your help!

pelwell, 2022-04-01:
There should be an official release next week.

macmpi, 2022-04-01:
@pelwell would you expect Synaptics to finally make such release in upstream linux-firmware, or would that be only from Raspberry Pi Ltd at this stage?
Did you manage to get an update on the License file as per https://github.com/raspberrypi/linux/issues/4723#issuecomment-984418504
Thanks.

pelwell, 2022-04-01:
I don't expect this release to change anything about the way Synaptics choose to distribute (or not) their firmwares.

The Licence issue is still ongoing - they have provided some words, but they make it sound as if people are not free to redistribute the firmware, which they are. It's almost as if lawyers get paid by the hour.

pelwell, 2022-04-11:
There's an official release from Synaptics available here: https://github.com/RPi-Distro/firmware-nonfree/blob/fdaf74c780ca7a29b12d62e5b0d37c38c2321e20/debian/config/brcm80211/brcm/brcmfmac43436-sdio.bin
This will become part of the next apt package release unless a regression is found.

Krautmaster, 2022-06-21:
hmm i have a pi zero 2 and the wifi reconnect is broken since my last update. Can it be related? As soon as it looses contect to my mesh its gone and it was working fine before
[image]

Krautmaster, 2022-06-25:
for methe new / current FW looses connection and keeps crashing:

> Jun 25 11:43:53 pi3a avahi-daemon[300]: New relevant interface wlan0.IPv4 for mDNS.
Jun 25 11:43:53 pi3a avahi-daemon[300]: Registering new address record for 169.254.47.171 on wlan0.IPv4.
Jun 25 11:43:55 pi3a ntpd[811]: Listen normally on 6 wlan0 169.254.47.171:123
Jun 25 11:43:55 pi3a ntpd[811]: new interface(s) found: waking up resolver
Jun 25 11:43:56 pi3a dhcpcd[570]: wlan0: no IPv6 Routers available
Jun 25 11:47:28 pi3a kernel: [ 544.751249] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Jun 25 11:47:28 pi3a kernel: [ 544.835372] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jun 25 11:47:28 pi3a kernel: [ 544.835399] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Jun 25 11:47:28 pi3a kernel: [ 545.275350] ieee80211 phy0: brcmf_netdev_start_xmit: xmit rejected state=0
Jun 25 11:47:28 pi3a avahi-daemon[300]: Interface wlan0.IPv6 no longer relevant for mDNS.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Leaving mDNS multicast group on interface wlan0.IPv6 with address fe80::9470:7258:3ba7:df8d.
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: carrier lost
Jun 25 11:47:28 pi3a avahi-daemon[300]: Interface wlan0.IPv4 no longer relevant for mDNS.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 169.254.47.171.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Withdrawing address record for fe80::9470:7258:3ba7:df8d on wlan0.
Jun 25 11:47:28 pi3a avahi-daemon[300]: Withdrawing address record for 169.254.47.171 on wlan0.
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: deleting address fe80::9470:7258:3ba7:df8d
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: deleting route to 169.254.0.0/16
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: deleting default route
Jun 25 11:47:28 pi3a systemd[1]: Starting Load/Save RF Kill Switch Status...
Jun 25 11:47:28 pi3a systemd[1]: Started Load/Save RF Kill Switch Status.
Jun 25 11:47:28 pi3a kernel: [ 545.516943] brcmfmac: brcmf_sdiod_probe: Failed to set F1 blocksize
Jun 25 11:47:28 pi3a kernel: [ 545.516996] brcmfmac: brcmf_sdio_bus_reset: Failed to probe after sdio device reset: ret -123
Jun 25 11:47:28 pi3a kernel: [ 545.517486] mmc1: card 0001 removed
Jun 25 11:47:28 pi3a dhcpcd[570]: wlan0: removing interface
Jun 25 11:47:28 pi3a dhcpcd-run-hooks[1163]: wlan0: stopping wpa_supplicant

Krautmaster, 2022-06-25:
debug log

> Jun 25 13:05:35 raspberrypi kernel: [ 1557.643416] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x393 tid 0
Jun 25 13:05:35 raspberrypi kernel: [ 1557.873367] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x3f7 tid 0
Jun 25 13:05:35 raspberrypi kernel: [ 1557.923328] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x40a tid 0
Jun 25 13:05:36 raspberrypi kernel: [ 1558.406358] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x446 tid 0
Jun 25 13:05:36 raspberrypi kernel: [ 1558.406392] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x44a tid 0
Jun 25 13:06:43 raspberrypi kernel: [ 1625.256876] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306299] brcmfmac: CONSOLE:
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306330] brcmfmac: CONSOLE: FWID 01-623162e
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306343] brcmfmac: CONSOLE: flags 1
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306355] brcmfmac: CONSOLE:
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306370] brcmfmac: CONSOLE: TRAP 3(6ff7c): pc 837364, lr 8372bb, sp 3f828, cpsr 40000003, spsr 41000000
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306385] brcmfmac: CONSOLE:   r0 0, r1 4, r2 0, r3 1010017, r4 0, r5 cc4, r6 60cbc
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306400] brcmfmac: CONSOLE:   r7 52868, r8 61190, r9 0, r10 1b9, r11 0, r12 7c
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306413] brcmfmac: CONSOLE:
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306426] brcmfmac: CONSOLE:    sp+0 00000000 00802fa9 00060de8 00061218
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306440] brcmfmac: CONSOLE:   sp+10 00065e00 00061190 00065e00 00060cbc
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306453] brcmfmac: CONSOLE:
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306465] brcmfmac: CONSOLE: sp+4 00802fa9
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306478] brcmfmac: CONSOLE: sp+34 00019ab5
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306490] brcmfmac: CONSOLE: sp+5c 00844f3f
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306503] brcmfmac: CONSOLE: sp+7c 008472b5
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306515] brcmfmac: CONSOLE: sp+94 0085e3f5
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306527] brcmfmac: CONSOLE: sp+fc 0081e9e9
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306540] brcmfmac: CONSOLE: sp+130 00844c95
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306552] brcmfmac: CONSOLE: sp+134 00844fe9
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306565] brcmfmac: CONSOLE: sp+164 0002f6df
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306577] brcmfmac: CONSOLE: sp+1bc 00807f53
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306590] brcmfmac: CONSOLE: sp+1f4 00875b39
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306602] brcmfmac: CONSOLE: sp+234 0001b6b1
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306614] brcmfmac: CONSOLE: sp+274 0001e085
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306627] brcmfmac: CONSOLE: sp+29c 0082eed3
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306639] brcmfmac: CONSOLE: sp+2c4 0082f215
Jun 25 13:06:43 raspberrypi kernel: [ 1625.306652] brcmfmac: CONSOLE: sp+2d4 0081f351
Jun 25 13:06:43 raspberrypi kernel: [ 1625.363199] ieee80211 phy0: brcmf_fil_cmd_data: bus is down. we have nothing to do.
Jun 25 13:06:43 raspberrypi kernel: [ 1625.363227] ieee80211 phy0: brcmf_cfg80211_get_tx_power: error (-5)
Jun 25 13:06:43 raspberrypi kernel: [ 1625.433318] ieee80211 phy0: brcmf_netdev_start_xmit: xmit rejected state=0
Jun 25 13:06:43 raspberrypi avahi-daemon[287]: Interface wlan0.IPv6 no longer relevant for mDNS.
Jun 25 13:06:43 raspberrypi avahi-daemon[287]: Leaving mDNS multicast group on interface wlan0.IPv6 with address 2a01:c22:c8c1:5d00:dc8:67bf:12a9:18c.
Jun 25 13:06:43 raspberrypi dhcpcd[628]: wlan0: carrier lost
Jun 25 13:06:43 raspberrypi avahi-daemon[287]: Interface wlan0.IPv4 no longer relevant for mDNS.
Jun 25 13:06:43 raspberrypi avahi-daemon[287]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 192.168.2.150.
Jun 25 13:06:43 raspberrypi avahi-daemon[287]: Withdrawing address record for 2a01:c22:c8c1:5d00:dc8:67bf:12a9:18c on wlan0.
Jun 25 13:06:43 raspberrypi avahi-daemon[287]: Withdrawing address record for 192.168.2.150 on wlan0.

pelwell, 2022-06-25:
Do you have `over_voltage=2` in config.txt? If not, please try it to see if the failure is power related (I suspect not, but want to be sure).

In order to report this mesh-related problem to Synaptics I'm going to need more details of the configuration - please open a new issue with as much information as you can.

Krautmaster, 2022-06-26:
yeah well i tried both, over_voltage 2 and 4, from my PI Zero Gen 1 i know its needed to get it stable streaming all day long.

Now the Version 1 ran stable all winter long but i did a rpi-update like a week go and suddenly it became instable. For me it looks like a mesh / wifi issue for now, but it similar to the instability i had with the Gen 1 pi Zero. A headless device is simply gone when it looses connection. 

For me it seems a bit related to FW right now. I mean thats whats get updated by rpi-update. And i even have the same with multiple mirco sds and OS setups for the same pi

Krautmaster, 2022-06-26:
Maybe the problem is caused by a bus issue. A script which is automatically rebooting the device keeps it accessible. Form syslog and messages it seems a bit like the bus get stuck or down at some point and that seem to lead to a wifi fw crash. The system itself stays  on. I think it's not really related to the wifi infrastructure as no other device (and I have a lot) has issues.

Krautmaster, 2022-06-27:
found no file with "f" so i did this:
[image]

after boot

> root@raspberrypi:/home/pi# sudo dmesg | grep brcmf
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=Composite-1:720x480@60i smsc95xx.macaddr=B8:27:EB:FD:BF:72 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=tty1 root=PARTUUID=fc6b844e-02 rootfstype=ext4 fsck.repair=yes rootwait brcmfmac.debug=0x100000
[    9.659498] brcmfmac: F1 signature read @0x18000000=0x1542a9a6
[    9.668485] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    9.677345] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
[    9.683064] usbcore: registered new interface driver brcmfmac
[    9.888719] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    9.888916] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
[    9.889084] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    9.889863] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270
[    9.995571] brcmfmac: CONSOLE:
[    9.995606] brcmfmac: CONSOLE: RTE (SDIO-CDC) 9.88.4.65 (test) (f149b32@shgit)  (r679549) on BCM43430 r2 @ 37.4/81.6/81.6MHz
[    9.995622] brcmfmac: CONSOLE: sdpcmdcdc0: Broadcom SDPCMD CDC driver
[    9.995637] brcmfmac: CONSOLE: reclaim section 0: Returned 50384 bytes to the heap
[    9.995650] brcmfmac: CONSOLE: GLICH crsmin 1
[    9.995665] brcmfmac: CONSOLE: wl0: Broadcom BCM43430 802.11 Wireless Controller 9.88.4.65 (test) (f149b32@shgit)  (r679549)
[    9.995679] brcmfmac: CONSOLE: wl_nd_ra_filter_init: Enter..
[    9.995692] brcmfmac: CONSOLE: TCAM: 256 used: 95 exceed:0
[    9.995706] brcmfmac: CONSOLE: reclaim section 1: Returned 101992 bytes to the heap
[    9.995720] brcmfmac: CONSOLE: threadX v5.6 initialized
[    9.995735] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[    9.995749] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :txbf
[    9.995762] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :rsdb_mode
[    9.995776] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :tdls_enable
[    9.995790] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :sup_wpa
[    9.995803] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :vhtmode
[    9.995817] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -29 :bw_cap
[    9.995832] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[    9.995847] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[   11.332828] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :toe_ol
[   11.332861] brcmfmac: CONSOLE: wl0: wl_open
[   11.357933] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   11.402794] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[   11.402827] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   11.402841] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   11.512845] brcmfmac: CONSOLE: wl0: wl_open
[   11.573110] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[   11.573143] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :vhtmode
[   11.932595] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   13.042713] brcmfmac: CONSOLE: wl0: link up (wl0)
[   13.042748] brcmfmac: CONSOLE: E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
[   13.082855] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 0
[   13.122893] brcmfmac: CONSOLE: E/GTK: wlc_bcol_process_gtk_rekey: 4 handshake frame, skip
[   13.122930] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   13.122944] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   13.206216] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   13.206245] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   13.303272] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   13.303352] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   14.602777] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   14.602811] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   18.462715] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 5
[   19.462788] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   19.462823] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 4
[   39.202570] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0xf8 tid 0
root@raspberrypi:/home/pi#

pelwell, 2022-06-27:
```
[ 9.889863] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct 9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit) (r679549) FWID 01-f40f3270
```
Yes, that's the version I was referring to.

If that ends up improving things it suggests a regression in the firmware, which bring me back to my request for a new issue with more details, i.e. what you did to a standard image, what you are running, e.t.c., so it can be reported to Synaptics.

Krautmaster, 2022-06-27:
actually it seems to be normal = stable now, like it had been working for me with stock bullseye on my pi zero 2. When i did the rpi-update i get the problematic latest version which seems to be stable if I stay on one AP but it seem to loose connection especially switching between the mesh APs. And then only a reboot seem to solve it. 

Will create a issue for that, thus im not completely able to say what might go wrong but i can provide logs and all.

Krautmaster, 2022-09-15:
i just ran update again and had the same issue again a few days. Now ill try a rpi update. 
My working workaround is:

wget https://github.com/RPi-Distro/firmware-nonfree/raw/d95f6614b19cb2d3ead4f2e2ea4ba0a7397ac5c0/debian/config/brcm80211/brcm/brcmfmac43436f-sdio.bin

 cp brcmfmac43436f-sdio.bin /lib/firmware/brcm/brcmfmac43436-sdio.bin

reboot

Krautmaster, 2022-09-15:
> Sep 14 17:33:41 raspberrypi kernel: [ 5179.926569] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Sep 14 17:33:42 raspberrypi kernel: [ 5181.030053] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Sep 14 17:33:47 raspberrypi wpa_action: creating sendsigs omission pidfile: /run/sendsigs.omit.d/wpasupplicant.wpa_supplicant.wlan0.pid
Sep 14 17:33:47 raspberrypi wpa_action: bssid=9c:c7:a6:67:6f:7d
Sep 14 17:33:47 raspberrypi wpa_action: freq=2412
Sep 14 17:33:47 raspberrypi wpa_action: ssid=M&M
Sep 14 17:33:47 raspberrypi wpa_action: id=0
Sep 14 17:33:47 raspberrypi wpa_action: mode=station
Sep 14 17:33:47 raspberrypi wpa_action: pairwise_cipher=CCMP
Sep 14 17:33:47 raspberrypi wpa_action: group_cipher=CCMP
Sep 14 17:33:47 raspberrypi wpa_action: key_mgmt=WPA2-PSK
Sep 14 17:33:47 raspberrypi wpa_action: wpa_state=COMPLETED
Sep 14 17:33:47 raspberrypi wpa_action: ip_address=192.168.2.150
Sep 14 17:33:47 raspberrypi wpa_action: p2p_device_address=e6:5f:01:58:07:f7
Sep 14 17:33:47 raspberrypi wpa_action: address=e4:5f:01:58:07:f7
Sep 14 17:33:47 raspberrypi wpa_action: uuid=ac2a6dc5-1b94-599d-88f9-294fbb515a71
Sep 14 17:33:47 raspberrypi wpa_action: WPA_IFACE=wlan0 WPA_ACTION=DISCONNECTED
Sep 14 17:33:47 raspberrypi wpa_action: WPA_ID=0 WPA_ID_STR= WPA_CTRL_DIR=/var/run/wpa_supplicant
Sep 14 17:33:47 raspberrypi wpa_action: ifdown wlan0
Sep 14 17:33:49 raspberrypi wpa_action: removing sendsigs omission pidfile: /run/sendsigs.omit.d/wpasupplicant.wpa_supplicant.wlan0.pid
Sep 14 17:33:49 raspberrypi kernel: [ 5188.800168] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Sep 14 17:33:50 raspberrypi wpa_action: WPA_IFACE=wlan0 WPA_ACTION=CONNECTED
Sep 14 17:33:50 raspberrypi wpa_action: WPA_ID=0 WPA_ID_STR= WPA_CTRL_DIR=/var/run/wpa_supplicant
Sep 14 17:33:50 raspberrypi wpa_action: ifup wlan0=default
Sep 14 17:33:50 raspberrypi kernel: [ 5188.881841] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
Sep 14 17:33:51 raspberrypi kernel: [ 5190.023289] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Sep 14 17:33:54 raspberrypi wpa_action: creating sendsigs omission pidfile: /run/sendsigs.omit.d/wpasupplicant.wpa_supplicant.wlan0.pid
Sep 14 17:33:54 raspberrypi wpa_action: bssid=9c:c7:a6:67:6f:7d
Sep 14 17:33:54 raspberrypi wpa_action: freq=2412
Sep 14 17:33:54 raspberrypi wpa_action: ssid=M&M
Sep 14 17:33:54 raspberrypi wpa_action: id=0
Sep 14 17:33:54 raspberrypi wpa_action: mode=station
Sep 14 17:33:54 raspberrypi wpa_action: pairwise_cipher=CCMP
Sep 14 17:33:54 raspberrypi wpa_action: group_cipher=CCMP
Sep 14 17:33:54 raspberrypi wpa_action: key_mgmt=WPA2-PSK
Sep 14 17:33:54 raspberrypi wpa_action: wpa_state=COMPLETED
Sep 14 17:33:54 raspberrypi wpa_action: ip_address=192.168.2.150
Sep 14 17:33:54 raspberrypi wpa_action: p2p_device_address=e6:5f:01:58:07:f7
Sep 14 17:33:54 raspberrypi wpa_action: address=e4:5f:01:58:07:f7
Sep 14 17:33:54 raspberrypi wpa_action: uuid=ac2a6dc5-1b94-599d-88f9-294fbb515a71
Sep 14 17:33:54 raspberrypi wpa_action: WPA_IFACE=wlan0 WPA_ACTION=DISCONNECTED
Sep 14 17:33:54 raspberrypi wpa_action: WPA_ID=0 WPA_ID_STR= WPA_CTRL_DIR=/var/run/wpa_supplicant
Sep 14 17:33:54 raspberrypi wpa_action: DISCONNECTED event blocked by hysteresis check
Sep 14 17:33:54 raspberrypi wpa_action: WPA_IFACE=wlan0 WPA_ACTION=CONNECTED
Sep 14 17:33:54 raspberrypi wpa_action: WPA_ID=0 WPA_ID_STR= WPA_CTRL_DIR=/var/run/wpa_supplicant
Sep 14 17:33:54 raspberrypi wpa_action: CONNECTED event blocked by hysteresis check
Sep 14 17:49:24 raspberrypi wpa_action: WPA_IFACE=wlan0 WPA_ACTION=DISCONNECTED
Sep 14 17:49:24 raspberrypi wpa_action: WPA_ID=0 WPA_ID_STR= WPA_CTRL_DIR=/var/run/wpa_supplicant
Sep 14 17:49:24 raspberrypi wpa_action: ifdown wlan0
Sep 14 17:49:25 raspberrypi kernel: [ 6124.077696] mmc1: card 0001 removed
Sep 14 17:49:25 raspberrypi kernel: [ 6124.191019] mmc1: queuing unknown CIS tuple 0x2c [20 f7 99 82 97 09 0c 00 ff 4b 72 8f e4 04 00 8a c4 20 ff f2 49 d0 ff 00 fa 26 46 d4 18 d9 88 e9 36 ff 0d 0e 14 ff 8c a8 11 9d 2c 4c 27 8e f7 00 0e 6c 18 ff 83 ee 1b 04 bb ff 35 ff 2e ae 00 1f] (207 bytes)
Sep 14 17:49:25 raspberrypi kernel: [ 6124.218176] mmc1: queuing unknown CIS tuple 0x69 [03 60 d6 4e 83 13 ff cf 5b 4b 7a 40 b7 08 b8 44 5d 26 7c a2 ce 44 ff 30 48 27 94 14 e5 30 e8 8d ff 04 e4 39 ec 19 30 e7 f6 65 fa 68 39 c5 0d 02 2f ff 99 3a 38 b2 f6 3a 2c 38 dc 1b c4 00 ef 40] (71 bytes)
Sep 14 17:49:25 raspberrypi kernel: [ 6124.271072] mmc1: queuing unknown CIS tuple 0x0b [6d ad ff 0b ff 39 a0 f0 2d d4 f9 c2 81 cc 40 72 a1 90 10 e8 00 01 10 10 41 a0 c9 3c ff 3a 57 72 e0 ac 0e 3c a5 9b 4c e3 27 be 9c 42 0d a0 f5 ca ac 91 4d 90 a4 58 22 02 01 ca 30 00 7a 82 0c 49] (153 bytes)
Sep 14 17:50:04 raspberrypi kernel: [ 6163.638274] mmc1: queuing unknown CIS tuple 0x2c [20 f7 99 82 97 09 0c 00 ff 4b 72 8f e4 04 00 8a c4 20 ff f2 49 d0 ff 00 fa 26 46 d4 18 d9 88 e9 36 ff 0d 0e 14 ff 8c a8 11 9d 2c 4c 27 8e f7 00 0e 6c 18 ff 83 ee 1b 04 bb ff 35 ff 2e ae 00 1f] (207 bytes)
Sep 14 17:50:04 raspberrypi kernel: [ 6163.668029] mmc1: queuing unknown CIS tuple 0x69 [03 60 d6 4e 83 13 ff cf 5b 4b 7a 40 b7 08 b8 44 5d 26 7c a2 ce 44 ff 30 48 27 94 14 e5 30 e8 8d ff 04 e4 39 ec 19 30 e7 f6 65 fa 68 39 c5 0d 02 2f ff 99 3a 38 b2 f6 3a 2c 38 dc 1b c4 00 ef 40] (71 bytes)
[... cut ...]  i2c_mux snd_soc_rpi_simple_soundcard snd_soc_bcm2835_i2s snd_soc_pcm5102a bcm2835_unicam snd_soc_core v4l2_dv_timings v4l2_fwnode v4l2_async bcm2835_v4l2(C) videobuf2_vmalloc bcm2835_codec(C) v4l2_mem2mem snd_compress snd_pcm_dmaengine bcm2835_isp(C) snd_pcm bcm2835_mmal_vchiq(C) videobuf2_dma_contig i2c_bcm2835 videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc snd_timer vc_sm_cma(C) snd uio_pdrv_genirq uio fixed drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6

pelwell, 2022-09-15:
Can you post the output of `dmesg | grep brcmfmac` for the two cases - with and without overwriting the supplied firmware?

Krautmaster, 2023-03-18:
sorry for the delay. Grabed my mower again for season start and did a full update ink rpi updateon the Pi Zero 2. The problem still exists. Wifi working fine for a few minutes and then suddenly gone

Log on fault:

> Mar 18 15:17:09 raspberrypi dhcpcd[434]: wlan0: requesting DHCPv6 information
Mar 18 15:17:09 raspberrypi avahi-daemon[364]: Withdrawing address record for fe80::3190:3a59:64ce:3b1c on wlan0.
Mar 18 15:17:09 raspberrypi dhcpcd[434]: wlan0: adding default route via fe80::52e6:36ff:febe:41c8
Mar 18 15:17:09 raspberrypi kernel: [   13.301738] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
Mar 18 15:17:10 raspberrypi kernel: [   13.501718] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 4
Mar 18 15:17:10 raspberrypi avahi-daemon[364]: Got SIGHUP, reloading.
Mar 18 15:17:10 raspberrypi avahi-daemon[364]: No service file found in /etc/avahi/services.
Mar 18 15:17:10 raspberrypi ntpd[648]: bind(24) AF_INET6 2a01:c23:7d0a:100:6369:9e0c:4ba4:d003#123 flags 0x11 failed: Cannot assign requested address
Mar 18 15:17:10 raspberrypi ntpd[648]: unable to create socket on wlan0 (6) for 2a01:c23:7d0a:100:6369:9e0c:4ba4:d003#123
Mar 18 15:17:10 raspberrypi ntpd[648]: failed to init interface for address 2a01:c23:7d0a:100:6369:9e0c:4ba4:d003
Mar 18 15:17:10 raspberrypi ntpd[648]: Listen normally on 7 wlan0 [fe80::3190:3a59:64ce:3b1c%2]:123
Mar 18 15:17:10 raspberrypi ntpd[648]: new interface(s) found: waking up resolver
Mar 18 15:17:13 raspberrypi dhcpcd[434]: wlan0: using static address 192.168.2.150/24
Mar 18 15:17:13 raspberrypi avahi-daemon[364]: Registering new address record for 192.168.2.150 on wlan0.IPv4.
Mar 18 15:17:13 raspberrypi dhcpcd[434]: wlan0: adding route to 192.168.2.0/24
Mar 18 15:17:13 raspberrypi dhcpcd[434]: wlan0: adding default route via 192.168.2.10
Mar 18 15:17:13 raspberrypi dhcpcd[434]: wlan0: pid 434 deleted route to 192.168.2.0/24
Mar 18 15:17:13 raspberrypi dhcpcd[434]: wlan0: pid 434 deleted default route via 192.168.2.10
Mar 18 15:17:14 raspberrypi kernel: [   18.354167] uart-pl011 3f201000.serial: no DMA platform data
Mar 18 15:17:14 raspberrypi ntpd[648]: Listen normally on 8 wlan0 192.168.2.150:123
Mar 18 15:17:14 raspberrypi ntpd[648]: bind(26) AF_INET6 2a01:c23:7d0a:100:6369:9e0c:4ba4:d003#123 flags 0x11 failed: Cannot assign requested address
Mar 18 15:17:14 raspberrypi ntpd[648]: unable to create socket on wlan0 (9) for 2a01:c23:7d0a:100:6369:9e0c:4ba4:d003#123
Mar 18 15:17:14 raspberrypi ntpd[648]: failed to init interface for address 2a01:c23:7d0a:100:6369:9e0c:4ba4:d003
Mar 18 15:17:14 raspberrypi ntpd[648]: new interface(s) found: waking up resolver
Mar 18 15:29:33 raspberrypi systemd[1]: Starting GPS (Global Positioning System) Daemon...
Mar 18 15:29:33 raspberrypi systemd[1]: Started GPS (Global Positioning System) Daemon.
Mar 18 15:29:36 raspberrypi ntpd[648]: Listen normally on 10 wlan0 [2a01:c23:7d0a:100:6369:9e0c:4ba4:d003]:123
Mar 18 15:29:36 raspberrypi ntpd[648]: 192.168.2.10 local addr 192.168.2.111 -> 192.168.2.150
Mar 18 15:29:36 raspberrypi ntpd[648]: new interface(s) found: waking up resolver
Mar 18 15:29:37 raspberrypi CRON[390]: (CRON) info (No MTA installed, discarding output)
Mar 18 15:29:47 raspberrypi systemd[1]: systemd-fsckd.service: Succeeded.
Mar 18 15:30:04 raspberrypi systemd[1]: Created slice User Slice of UID 1000.
Mar 18 15:30:04 raspberrypi systemd[1]: Starting User Runtime Directory /run/user/1000...
Mar 18 15:30:04 raspberrypi systemd[1]: Finished User Runtime Directory /run/user/1000.
Mar 18 15:30:04 raspberrypi systemd[1]: Starting User Manager for UID 1000...
Mar 18 15:30:05 raspberrypi systemd[906]: Queued start job for default target Main User Target.
Mar 18 15:30:05 raspberrypi systemd[906]: Created slice User Application Slice.
Mar 18 15:30:05 raspberrypi systemd[906]: Reached target Paths.
Mar 18 15:30:05 raspberrypi systemd[906]: Reached target Timers.
Mar 18 15:30:05 raspberrypi systemd[906]: Starting D-Bus User Message Bus Socket.
Mar 18 15:30:05 raspberrypi systemd[906]: Listening on GnuPG network certificate management daemon.
Mar 18 15:30:05 raspberrypi systemd[906]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Mar 18 15:30:05 raspberrypi systemd[906]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Mar 18 15:30:05 raspberrypi systemd[906]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Mar 18 15:30:05 raspberrypi systemd[906]: Listening on GnuPG cryptographic agent and passphrase cache.
Mar 18 15:30:05 raspberrypi systemd[906]: Listening on D-Bus User Message Bus Socket.
Mar 18 15:30:05 raspberrypi systemd[906]: Reached target Sockets.
Mar 18 15:30:05 raspberrypi systemd[906]: Reached target Basic System.
Mar 18 15:30:05 raspberrypi systemd[1]: Started User Manager for UID 1000.
Mar 18 15:30:05 raspberrypi systemd[1]: Started Session 2 of user pi.
Mar 18 15:30:05 raspberrypi systemd[906]: Reached target Main User Target.
Mar 18 15:30:05 raspberrypi systemd[906]: Startup finished in 671ms.

what I did now was:

`wget https://github.com/RPi-Distro/firmware-nonfree/raw/d95f6614b19cb2d3ead4f2e2ea4ba0a7397ac5c0/debian/config/brcm80211/brcm/brcmfmac43436f-sdio.bin`
` cp brcmfmac43436f-sdio.bin /lib/firmware/brcm/brcmfmac43436-sdio.bin`

> root@raspberrypi:/home/pi# dmesg | grep brcmfmac
[... cut ...]
[    9.772283] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :tdls_sta_info
[    9.842345] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[    9.842368] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[    9.902289] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   10.112337] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   11.552207] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 4
[   11.652219] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[   11.652237] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe

raspiduino, 2024-03-19:
I would like to say a huge thank you to the person created this issue. It may seem unrelated, but I have a set-top box based on S805X SoC with AP6236A WiFi chip. It's detected by Linux as BCM43430/2 and I tried to download the firmware for BCM43430 but it does not work. Until I found this issue and realized that BCM43430/2 needs BCM43436's firmware and now it worked. Still don't understand why they need BCM43436's firmware for BCM43430/2 and I hate their naming system!

pelwell, 2024-03-19:
> Still don't understand why they need BCM43436's firmware for BCM43430/2 and I hate their naming system!

BCM43430/2 can be thought of as "variant 2 of the BCM43430" family, which is called BCM43436 (or perhaps SYN43436) by Synaptics. I'm not a fan, either.

raspiduino, 2024-03-19:
Btw now the firmware loaded without error, and wlan0 is shown up in `ifconfig`, but when I tried scanning APs using `iwlist scan`, it always say "no result". Also, `ifup` and `ipdown` always return error, with `ifup` claiming wlan0 is not configured. Can someone help?

raspiduino, 2024-08-02:
Okay I found the solution. Just use latest bcm43430b0 firmware from armbian and you will get it working. Again I hate their naming system.