RPi-Distro/firmware-nonfree #24: Distro guidance for how to "install" the new rpi02w firmware

colemickens, 2022-05-28:
Hi,

I thought I had understood the gist of #23, but I'm still not seeing any wireless devices on the rpi02w.

I'm linking to the raw Nix file, but hopefully it's pretty explicit about what I'm doing:

https://github.com/colemickens/nixpkgs/blob/ca30f6b48d4be7a83fc479a4838d08d663f1146b/pkgs/os-specific/linux/firmware/raspberrypi-wireless/default.nix#L45-L50

```
    ln -s "./${rpi02w}-sdio.txt" "$out/lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2.txt"
    ln -s "./${rpi02w}-sdio.txt" "$out/lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt"
    ln -s "./${rpi02w}f-sdio.txt" "$out/lib/firmware/brcm/brcmfmac43436f-sdio.raspberrypi,model-zero-2.txt"
    ln -s "./${rpi02w}f-sdio.txt" "$out/lib/firmware/brcm/brcmfmac43436f-sdio.raspberrypi,model-zero-2-w.txt"
    ln -s "./${rpi02w}s-sdio.txt" "$out/lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2.txt"
    ln -s "./${rpi02w}s-sdio.txt" "$out/lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt"
```

where `rpi02w` is set to `brcmfmac43436`.

Questions:

* What is the "device name" that is used for the text file here (is this in device tree maybe?) Do I symlink all three variants of the firmware?
* Is the `rpi02w` value correct?
* What else should I check?

Thanks

colemickens, 2022-05-30:
I got something to work:

https://github.com/colemickens/nixpkgs/commit/f94e62aaa2c0223e3c08259515b62b208161fc2b

XECDesign, 2022-05-31:
This kind of thing is probably better addressed on the forum.

I'd just say compare what we ship Raspberry Pi OS with what ends up installed by the Nix package. Looking just as the .deb may be a bit misleading, since we use 'update-alternatives' in `postinst`.

closed 2022-05-31 by XECDesign

XECDesign, 2022-06-13:
Heads up that some file names have changed.

`cypress/cyfmac43455-sdio-standard.bin` should be installed, or linked to `brcm/brcmfmac43455-sdio.bin`
 You can see [here](https://github.com/RPi-Distro/firmware-nonfree/blob/bullseye/debian/config/brcm80211/cypress/README.txt) for the difference between the standard and minimal versions.

`brcm/brcmfmac43456f-sdio.bin` has been renamed to  `brcm/brcmfmac43456-sdio.bin` and should be installed as such.