RPi-Distro/firmware-nonfree #27: Raspberry Pi Zero 2 W: BCM43430/2 firmware stuck when switching between mesh endpoints

Krautmaster, 2022-06-27:
I was running "rpi-update" on my Pi Zero 2 and it seems like its not able to reconnect to the wifi when it is one once. My Pi is driving over my place and switching between three AP with a AVM mesh. 

Problematic FW:
> Jun 27 10:45:11 raspberrypi kernel: [    9.153090] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
Jun 27 10:45:11 raspberrypi kernel: [    9.153436] usbcore: registered new interface driver brcmfmac
Jun 27 10:45:11 raspberrypi kernel: [    9.159229] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.bin failed with error -2
Jun 27 10:45:11 raspberrypi kernel: [    9.189398] vc4-drm soc:gpu: bound 3f400000.hvs (ops vc4_hvs_ops [vc4])
Jun 27 10:45:11 raspberrypi kernel: [    9.198736] Registered IR keymap rc-cec
Jun 27 10:45:11 raspberrypi kernel: [    9.199702] rc rc0: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0
Jun 27 10:45:11 raspberrypi kernel: [    9.201039] input: vc4 as /devices/platform/soc/3f902000.hdmi/rc/rc0/input1
Jun 27 10:45:11 raspberrypi kernel: [    9.364497] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
Jun 27 10:45:11 raspberrypi kernel: [    9.364722] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43436-sdio for chip BCM43430/2
Jun 27 10:45:11 raspberrypi kernel: [    9.364899] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
Jun 27 10:45:11 raspberrypi kernel: [    9.365682] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Mar 29 2022 19:22:43 version 9.88.4.77 (g58bc5cc) FWID 01-623162e
Jun 27 10:45:11 raspberrypi kernel: [    9.414477] brcmfmac: CONSOLE:
Jun 27 10:45:11 raspberrypi kernel: [    9.414518] brcmfmac: CONSOLE: RTE (SDIO-CDC) 9.88.4.77 on BCM43430 r2 @ 37.4/81.6/81.6MHz
Jun 27 10:45:11 raspberrypi kernel: [    9.414534] brcmfmac: CONSOLE: sdpcmdcdc0: Broadcom SDPCMD CDC driver
Jun 27 10:45:11 raspberrypi kernel: [    9.414548] brcmfmac: CONSOLE: reclaim section 0: Returned 49968 bytes to the heap
Jun 27 10:45:11 raspberrypi kernel: [    9.414561] brcmfmac: CONSOLE: GLICH crsmin 1
Jun 27 10:45:11 raspberrypi kernel: [    9.414576] brcmfmac: CONSOLE: wl0: Broadcom BCM43430 802.11 Wireless Controller 9.88.4.77 (g58bc5cc)
Jun 27 10:45:11 raspberrypi kernel: [    9.414590] brcmfmac: CONSOLE: wl_nd_ra_filter_init: Enter..
Jun 27 10:45:11 raspberrypi kernel: [    9.414603] brcmfmac: CONSOLE: TCAM: 256 used: 98 exceed:0
Jun 27 10:45:11 raspberrypi kernel: [    9.414617] brcmfmac: CONSOLE: reclaim section 1: Returned 102716 bytes to the heap
Jun 27 10:45:11 raspberrypi kernel: [    9.414631] brcmfmac: CONSOLE: threadX v5.6 initialized
Jun 27 10:45:11 raspberrypi kernel: [    9.414645] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
Jun 27 10:45:11 raspberrypi kernel: [    9.414659] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :txbf
Jun 27 10:45:11 raspberrypi kernel: [    9.414673] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :rsdb_mode
Jun 27 10:45:11 raspberrypi kernel: [    9.414687] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :tdls_enable
Jun 27 10:45:11 raspberrypi kernel: [    9.414701] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :vhtmode
Jun 27 10:45:11 raspberrypi kernel: [    9.414714] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -29 :bw_cap
Jun 27 10:45:11 raspberrypi kernel: [    9.414729] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
Jun 27 10:45:11 raspberrypi kernel: [    9.414745] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
Jun 27 10:45:11 raspberrypi kernel: [   10.233359] 8021q: 802.1Q VLAN Support v1.8
Jun 27 10:45:11 raspberrypi rsyslogd: imuxsock: Acquired UNIX socket '/run/systemd/journal/syslog' (fd 3) from systemd.  [v8.2102.0]
Jun 27 10:45:11 raspberrypi rsyslogd: [origin software="rsyslogd" swVersion="8.2102.0" x-pid="317" x-info="https://www.rsyslog.com"] start
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Successfully called chroot().
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Successfully dropped remaining capabilities.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: No service file found in /etc/avahi/services.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: New relevant interface lo.IPv6 for mDNS.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: New relevant interface lo.IPv4 for mDNS.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Network interface enumeration completed.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Registering new address record for ::1 on lo.*.
Jun 27 10:45:11 raspberrypi avahi-daemon[289]: Registering new address record for 127.0.0.1 on lo.IPv4.
Jun 27 10:45:11 raspberrypi CRON[371]: (root) CMD (~/start.sh)

Syslog
https://cloud.krautmaster.de/index.php/s/esWJoemFiKzpqys

Moving back to FW:

> 9.889863] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Oct  9 2020 14:44:32 version 9.88.4.65 (test) (f149b32@shgit)  (r679549) FWID 01-f40f3270

seem to fix the problem of loosing the ability to do a auto reconnect and switches normally between the APs in the mesh with ony a short outage.

Krautmaster, 2022-06-27:
need to check the same. But there is no need to update right now. I was only noticing that it was running fine and the last rpi-update seem to have broken my wifi functionality in the mentioned aspect.

I can run rpi-update again, it tells me it can update but i'd expect it will pull the problematic 9.88.4.77 again (which i manually replaced with 9.88.4.65 currently).

For further tests and if its needed i can update again to provide more details in debugging and so on. Shall I?

pelwell, 2022-06-27:
`rpi-update` _will not_ update your WiFi firmware - `apt update` might.