RPi-Distro/firmware-nonfree #28: Possible redundant section in the copyright file or mistake

agherzan, 2022-07-27:
The copyright file currently partitions the files provided in this repo as follows:

```
Files: brcm/bcm43xx*.fw brcm/brcmfmac*.bin
Copyright: Broadcom Corporation
License: binary-redist-Broadcom-wifi

Files: brcm/BCM-0bb4-0306.hcd debian/config/brcm80211/*/*43430*
 debian/config/brcm80211/*/*43455*
Copyright: 2018, Cypress Semiconductor Corporation
License: binary-redist-Cypress

Files: debian/config/brcm80211/*/*43436* debian/config/brcm80211/*/*43456*
Copyright: Synaptics
License: Synaptics
```

Now, matching this to actual files, we end up with:

```
ls -al ../../../debian/config/brcm80211/brcm/bcm43xx*.fw ../../../debian/config/brcm80211/brcm/brcmfmac*.bin

ls: cannot access '../../../debian/config/brcm80211/brcm/bcm43xx*.fw': No such file or directory
debian/config/brcm80211/brcm/brcmfmac43430-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43436-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43455-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43456-sdio.bin

ls -al brcm/BCM-0bb4-0306.hcd ../../../debian/config/brcm80211/*/*43430*  ../../../debian/config/brcm80211/*/*43455*

ls: cannot access 'brcm/BCM-0bb4-0306.hcd': No such file or directory
debian/config/brcm80211/brcm/brcmfmac43430-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43430-sdio.clm_blob
debian/config/brcm80211/brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt
debian/config/brcm80211/brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt
debian/config/brcm80211/brcm/brcmfmac43430-sdio.txt
debian/config/brcm80211/brcm/brcmfmac43455-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43455-sdio.clm_blob
debian/config/brcm80211/brcm/brcmfmac43455-sdio.raspberrypi,3-model-a-plus.txt
debian/config/brcm80211/brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt
debian/config/brcm80211/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.txt
debian/config/brcm80211/brcm/brcmfmac43455-sdio.txt
debian/config/brcm80211/cypress/cyfmac43430-sdio.bin
debian/config/brcm80211/cypress/cyfmac43430-sdio.clm_blob
debian/config/brcm80211/cypress/cyfmac43455-sdio.clm_blob
debian/config/brcm80211/cypress/cyfmac43455-sdio-minimal.bin
debian/config/brcm80211/cypress/cyfmac43455-sdio-standard.bin

ls -al ../../../debian/config/brcm80211/*/*43436* ../../../debian/config/brcm80211/*/*43456*
debian/config/brcm80211/brcm/brcmfmac43436-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43436-sdio.clm_blob
debian/config/brcm80211/brcm/brcmfmac43436-sdio.txt
debian/config/brcm80211/brcm/brcmfmac43436s-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43436s-sdio.txt
debian/config/brcm80211/brcm/brcmfmac43456-sdio.bin
debian/config/brcm80211/brcm/brcmfmac43456-sdio.clm_blob
debian/config/brcm80211/brcm/brcmfmac43456-sdio.txt
```

As you can see, the files in the first section, `Copyright: Broadcom Corporation`, are all included later in either `Copyright: 2018, Cypress Semiconductor Corporation` or `Copyright: Synaptics`.

Assuming that this is meant to be based/using the Debian copyright file format (https://dep-team.pages.debian.net/deps/dep5/#files), it looks like the first paragraph is redundant because:

    Multiple Files paragraphs are allowed. The last paragraph that matches a particular file applies to it.

So given that all the files matched by the first paragraph are matched by later defined paragraphs, makes the first paragraph, in theory, either redundant, a mistake or the assumptions here are wrong (file format, semantics, etc).

@pelwell would you be able to clarify this? I think we should also do the same in the actual `copyright` file as I would expect that it is a source of confusion at this point.

XECDesign, 2022-07-29:
In your first `ls` command, where did `debian/config/brcm80211` come from?

Once the upstream tarball is extracted, the first paragraph will apply to files which aren't covered by later licenses.

agherzan, 2022-07-29:
So using this repository (https://github.com/RPi-Distro/firmware-nonfree, no upstream tarball), the first files `copyright` statement is: `Files: brcm/bcm43xx*.fw brcm/brcmfmac*.bin`. This matches the following files:

```
brcm/brcmfmac43430-sdio.bin
brcm/brcmfmac43436-sdio.bin
brcm/brcmfmac43436s-sdio.bin
brcm/brcmfmac43455-sdio.bin
brcm/brcmfmac43456-sdio.bin
```

As an example, the second files copyright statement is: `Files: brcm/BCM-0bb4-0306.hcd debian/config/brcm80211/*/*43430*`. This matches the following files:

```
brcm/brcmfmac43430-sdio.bin
brcm/brcmfmac43430-sdio.clm_blob
brcm/brcmfmac43430-sdio.raspberrypi,3-model-b.txt
brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt
brcm/brcmfmac43430-sdio.txt
cypress/cyfmac43430-sdio.bin
cyfmac43430-sdio.clm_blob
```

In this example, `brcm/brcmfmac43430-sdio.bin` is matched in both cases.

As an extra note, it seems like the copyright file uses both relative paths (to the file itself) and relative paths to the root of the repository. Is there any benefit in that? Or maybe that's the root of this confusion.

XECDesign, 2022-07-29:
All paths are relative to the root of the repo.

agherzan, 2022-07-29:
@XECDesign if that is the case, what are these paths pointing to? 

* brcm/bcm43xx*.fw
* brcm/brcmfmac*.bin
* brcm/BCM-0bb4-0306.hcd

Those paths don't exist at the root of the repo.

XECDesign, 2022-07-29:
In this case, it refers to the files from the upstream tarball. The notion is inherited from debian's packaging: https://salsa.debian.org/kernel-team/firmware-nonfree/-/blob/master/debian/copyright

agherzan, 2022-07-29:
That is clear now. So basically, when using this repository directly (not in conjunction with the upstream tarball), the first section doesn't match anything.

closed 2022-07-29 by agherzan