RPi-Distro/firmware-nonfree #29: Possible legal risk as part of the Synaptics license terms

agherzan, 2022-07-27:
This repository has included lately the full test of the `Synaptics` license as part of the `copyright` file clarifying in this way the full terms of the respective license. In general, the text looks pretty standard - redistribution rights for use solely in connection with specific products. The following clause though appears to be problematic:

> 10. The term of this agreement (the ?License Term?) will begin on the date of accessing, using, downloading or installing the Software Libraries and will end when you stop accessing or using the Software Libraries or when Synaptics notifies you in writing of the termination of this agreement not less than thirty days following the sending of the notice, unless this agreement is terminated sooner by Synaptics as provided herein. Synaptics has the right, in its sole discretion, to terminate this agreement immediately by giving written notice of termination to you. If you breach any of the provisions contained within this agreement, the license and agreement shall automatically and immediately terminate and Synaptics shall seek all remedies available to it at law or in equity.

I have copied the entire clause for context but the concern is especially about the following: `Synaptics has the right, in its sole discretion, to terminate this agreement immediately by giving written notice of termination to you.`. This can pose a legal risk as it can behave (in legal terms) as a **killswitch**. In other words, there is no certainty that the license will remain valid during the lifespan of a product and that the rug will not be pulled behind the device maker’s feet any time soon, thereby frustrating the investment made.

I'm not a lawyer but this concern was raised by an IP team (internal to the projects I’m working on) and it can be as simple as a copy/paste mistake (in which case we can easily fix it) or as complicated as a confirmed risk that the downstream users should be aware of when deploying products that include the associated blobs.

I’m noting that the blobs in question affect the following boards:

- Raspberry Pi Zero 2W
- Raspberry Pi 4

pelwell, 2022-07-27:
Synaptics may reserve the right to use a killswitch, but it would be commercial suicide. I'm sure you could get reasonable terms on a business insurance policy against such an eventuality.

kappapiana, 2022-07-27:
Sorry to be blunt, but "you can get an insurance to cover" is not quite an acceptable answer to this kind of problems, not just in terms of costs (the margins in low consumption devices are slim), but also in terms of costs that cannot be easily covered or that insurances typically don't cover, like the reputational damage or similar. The fact that terminating a license would be commercial disaster so an unlikely event is not necessarily true either. A termination could be (and likely would be) individual, and the provider could have real reasons and ostensible reasons or even no reasons to terminate to one particular customer, maybe based on their nationality, or upon some strong hinting from their home government. I see that in this space termination is a last resort used as a means of retaliation in case of legal harassment or anyway with a good cause, not "just because".

pelwell, 2022-07-27:
When governments get involved the license agreement suddenly becomes rather irrelevant. International laws are there for the flouting.

agherzan, 2022-07-27:
What worries me is the fact that we transfer the risk to the users of the projects that take in these blobs. For example a build system BSP, a distro supporting these BSP blobs and so on. All downstream people/businesses need to be aware so that they can take a decision and assess the risk themselves. @pelwell, your answer here seems to imply the fact that this is a known issue/fact and everybody should have insurance to cover it. But that can only happen when everybody is aware of the risk (no matter how likely the risk is). So, if that is the case, how are the current Linux distributions (eg. Raspberry Pi OS) managing this communication to their users? Do people sign a EULA (of some sort) that makes them aware of the possible legal issues?

This is more of a practical question that affects technically my projects but @kappapiana's points are equally valid.

pelwell, 2022-07-29:
We're currently exploring the options.

Note that this licence only applies to Pi 4 in the sense that the Pi 400 is a member of the Pi 4 family.

agherzan, 2022-07-29:
Thanks @pelwell, appreciated.

madscientist42, 2023-09-05:
> Synaptics may reserve the right to use a killswitch, but it would be commercial suicide. I'm sure you could get reasonable terms on a business insurance policy against such an eventuality.

That rather doesn't mean they won't go there.  Do remember SCOX...

madscientist42, 2023-09-05:
One has to wonder...with a project that CLAIMS to be for Educational uses and is heavy Open Source, etc. why they would be using a design that has this problem (And, YES, it is one...)

pelwell, 2023-11-06:
No, it's not a joke, and you're not imagining it, there is a new licence file from Synaptics - see here: https://github.com/RPi-Distro/firmware-nonfree/pull/40

madscientist42, 2023-11-06:
Geez...that's smoking some serious buttcrack there.  Does anyone know if this bleeds into the Pi5?

pelwell, 2023-11-06:
You're going to have to rephrase that if you want a meaningful answer.

agherzan, 2023-11-06:
Thanks @pelwell. @kappapiana the proposed updated above looks alright to me. What do you think?

madscientist42, 2023-11-06:
Heh.  Did they end up using similar tech needing the license in question?

pelwell, 2023-11-06:
Pi 5 shares the same WiFi part as the Pi 4 (not so different from 43456, but not identical), however the SDIO interface runs faster so the bandwidth is better.

The reason there's an updated licence is that we didn't give up, and eventually we got a new point of contact at Synaptics who understood what was needed and why, and who had the authority and persistence to extract it from the lawyers.

madscientist42, 2023-11-06:
Ok, thanks for the clarification. And, it's GOOD to hear...  I appreciate
the work- because I've done some of the same class of efforts.

alpianon, 2023-11-07:
@agherzan I shared views with @kappapiana. The termination clause looks much better now, since the explicit kill-switch clause is gone, at least. From a licensee perspective, it is still not perfect, because it does not explicitly say that the license is irrevocable unless terminated for breach of its terms. Based on general principles on perpetual obligations, shared across different jurisdictions, courts often find that permanent agreements may be terminated at will, by and large; but in this specific case the licensor has no positive obligations under the license, so IMO the conclusion may be different.
Since the laws of the State of California apply, only some legal expert from there may give a more concrete answer.

pelwell, 2023-11-07:
My unexpert reading is that if you are prepared to accept the licences for the other Pi-required firmwares here then this new Synpatics licence should be no more troubling.

alpianon, 2023-11-07:
To make it clearer, if the licence is not explicitly qualified as "perpetual" or "irrevocable", courts may find that it may be terminated at will, even without an explicit kill-switch clause. The expression "This Agreement is effective until terminated" may give leeway to this interpretation. But again, it depends on the specific aspect of the case, and on the applicable law.

pelwell, 2023-11-07:
In case anyone is under the wrong impression, this is not a "choose your own licence" exercise. Once the repo owner has had a chance to confirm that I've not messed up the formatting, the PR will be merged.

kappapiana, 2023-11-07:
> In case anyone is under the wrong impression, this is not a "choose your own licence" exercise. Once the repo owner has had a chance to confirm that I've not messed up the formatting, the PR will be merged.

Not sure I get what you mean by that. Our interjection is to clarify the point of view of two respected lawyers on whether this license is sane for a client to take, since it is us who have spotted the Killswitch in the first place and advised a very large client and an open source project to stay away from him barring its removal.  

So, in case anyone was under the wrong impression that the change did resolve our, or anyone's concerns, then we must set the record straight that it isn't quite the case, and why.

pelwell, 2023-11-07:
And are these two respected lawyers less concerned about the Cypress licence?

kappapiana, 2023-11-07:
Not taking offence from the implied sarcasm, we have encountered killswitches also in Cypress and we have highlighted them. The language is shaky at best, irrespective of who has conceived or is using it. Snide remarks won't change it.

Please note that "everybody does it as well" is not a great excuse and does not change our legal analysis. At best, it can cause other red flags to be raised that were not previously evident.

pelwell, 2023-11-07:
Closing now that #40 is merged.

closed 2023-11-07 by pelwell