RPi-Distro/firmware-nonfree #31: Raspberry Pi Zero 2 W - Bluetooth version 4.1.

microweeAlanP, 2022-09-19:
Hello Everyone,
I need to implement DLE (Data Length Extension) feature in my project. Basic idea is that Raspberry Pi behaves as a Central device that will send read requests to the Peripheral device. Peripheral device will respond with data that should be sent inside one packet. That is why I need DLE feature on both sides (Central and Peripheral). The problem is that Raspberry Pi Zero 2 W should have Bluetooth version 4.2 (first version where DLE was implemented), but from some reason I have Bluetooth version 4.1. (DLE in this version is not implemented).
I tested this on multiple ways. One of them was to check with hcitool and btmon. I used LE Read Local Supported Features Command to check what features does the controller on RP02W supports. This is the result that I got:
 
[image]

You can see that there is no LE Data Length Extension feature supported. Also I checked using hciconfig hci0 revision command where I saw that the version is 4.1.
 
[image]

In Raspberry Pi 4 specifications we have that it supports Bluetooth version 5.0 so I used Raspberry Pi 4 to compare the results and for RP4 I got that it supports DLE and that the version is the one from the specifications:
 
[image]

[image]

I tried to install multiple different versions of Raspbian and I have also tried to modify folder /lib/firmware/bcrm/ to try to load different firmware to the device but every time the result was the same. HCI version stayed the same. 
I have also done update and upgrade of the system and Bluez, as well as using other OS as Kali and Ubuntu server, but nothing changed.
Does someone know how to use Bluetooth 4.2 on Raspberry Pi Zero 2 W?

pelwell, 2022-09-20:
As I understand it, the Data Length extension (like the other features added in the v4.2 specification) is optional. Provided the mandatory bug fixes have been applied, manufacturers (Infineon, Synaptics, etc.) can rightly claim 4.2 compatibility without implementing all the new features.

microweeAlanP, 2022-09-23:
Thank you for your reply, @pelwell.
I would like to have version 4.2 . I have 3 Raspberry Pi Zero 2 W , but on all of them it is the same version, version 4.1.

pelwell, 2022-09-26:
We're waiting on a response from Synaptics about the 43436S firmware.

pelwell, 2022-09-28:
After a dialogue with Synaptics, we've identified that the current 43436S firmware incorrectly identifies itself - it should really say it supports Bluetooth 4.2 - but otherwise the feature set is correct. As I said earlier, BLE Data Length is an optional extension, and the 43436S does not support it.

The situation is complicated by the fact that, due to the inability of large chunks of the semiconductor industry to manufacture anything much at the moment, and the fact that we'd really like to be able to sell some Zero 2 Ws, there are two variants: one with 43436S and one with the slightly newer 43436P (that we refer to as just "43436"). Even without the cap on the module it is almost impossible to tell the two apart - they are pin-compatible, very small and very shiny - but if you get the light just right and have a camera with a macro lens you can just about read the laser-etched part number. Also, the manufacturing process doesn't record which part goes onto each unit, so really the only way to work out which part is fitted is to boot them up and check the kernel log to see which firmware is loaded.

With the 43436P being slightly newer it is able to support DLE, and the firmware does so. It also correctly reports Bluetooth 4.2. We are hoping to get a new firmware from Synaptics fairly soon, but unfortunately for you it won't change the capabilities of your Zero 2 Ws - just clarify the Bluetooth version.

danielprehealth, 2022-09-29:
Hmm thanks for sharing this, @pelwell.

Is there any plan to be able to distinguish the Zero 2 W’s with the 43436S from the Zero 2 W’s with the 43436P/43436 BLE chip in the future? Perhaps request production to add a little traceability on which chips go into which devices? We want to use the Zero 2 W in our product in which we aim to buy 10k+ Zero 2 W’s next year, and hopefully 100k+ in the following years. But we need the ability to confidently buy Zero 2 W’s that have the 43436P/43436 chip that has the DLE functionality. Is there someone we can reach out to at Raspberry Pi's Industry that might be able to help us with this procurement requirement?

pelwell, 2022-09-29:
The commercial/industrial department is telling me that, due to supply constraints of other components, the Zero 2 W is not suitable for new designs.

pelwell, 2022-09-29:
Our commercial director is happy for you to email him to discuss your requirements - Mike Buffham <mike.buffham@raspberrypi.com>.

fanoush, 2022-09-30:
Hello, this issue was discussed also here https://forums.raspberrypi.com/viewtopic.php?p=2040251#p2040251 and
I do have Zero 2W bought in early days after its release and it does report 4.2 and e.g. higher MTU negotiation works fine and I do get larger packets when connected from Zero to BLE device.

Are you saying newer Z2Ws are step back regarding BLE and do not support this anymore?
Or I got it wrong and it doesn't work the other way even with my Zero? I did not try negotiating higher MTU when connecting from other device to the Z2W.

pelwell, 2022-09-30:
I'm not sure about the production timeline, but you are correct that some Z2Ws can do DLE and others can't.

nickwinn, 2024-10-26:
Resurrecting a old thread as I just bought two new Raspberry Pi Zero 2 W from my local microcenter and it's reporting as a 4.1.

`┌──(student0341a㉿latchfruit)-[~]
└─$ hciconfig hci0 version revision
hci0:   Type: Primary  Bus: UART
        BD Address: B8:27:EB:C7:78:58  ACL MTU: 1021:8  SCO MTU: 64:1
        HCI Version: 4.1 (0x7)  Revision: 0x1fc
        LMP Version: 4.1 (0x7)  Subversion: 0x2209
        Manufacturer: Broadcom Corporation (15)
        Firmware 252.34 / 9
`

I bought these for a bluetooth security class I'm attending in the near future, syllabus says it requires 4.2. What's the deal here?

JamesH65, 2024-11-22:
@pelwell Did we ever get any information back on this? Our stack apparently still says 4.1.

fanoush, 2024-11-22:
BTW, it is not about version number itself, it is about chip features. Bluetooth LE without DLE (= you cannot increase MTU over 23) is horribly outdated now. If there was some issue in covid times about chip shortage and you temporarily used worse chip then OK, but shipping chip without DLE now and investing effort into increasing version from 4.1 to 4.2 just because DLE is optional is a waste of time on both sides - you will only confuse and annoy people that will expect DLE to work (as seen in first post).

pelwell, 2024-11-22:
The fact that some Z2Ws have 43436S and others 43436P hasn't changed (and can't, obviously - all that can be changed is the future part allocation). No amount of firmware effort will add DLE to the 43436S.

fanoush, 2024-11-22:
> No amount of firmware effort will add DLE to the 43436S.

Sure, what I tried to say that increasing the '1' byte in the firmware into '2'  for 43436S just because you can, because technically DLE is optional, won't help anyone, and will mostly cause more confusion and annoy people. It will only be harder to figure out which Zero is which and why higher MTU does not work. 

If you could change the allocation and stop producing Zero 2W with the 43436S (if  you still do) that would be of course much better, but it is probably off topic here.

pelwell, 2024-11-22:
Indeed, but I wasn't really addressing you.