RPi-Distro/firmware-nonfree #32: Unable to connect to WiFi on Raspberry Pi Zero 2 W (custom OS)

agners, 2022-12-22:
We are using the Raspberry Pi kernel and firmware to build Home Assistant OS, a custom built OS  using buildroot. In our HAOS 8.5 release we have been using `brcmfmac43436-sdio.bin` with md5sum 5dcbec4e28e90a4f503c1117fc71bdd9 (from commit https://github.com/RPi-Distro/firmware-nonfree/commit/dcea7a3, version 9.88.4.65). This seemed to have worked well.

With HAOS 9.x series we moved to `brcmfmac43436-sdio.bin` with md5sum c9ca5766578bcb603239de54737ede6a (from commit  https://github.com/RPi-Distro/firmware-nonfree/commit/28bad1e, version 9.88.4.77). With this version WiFi association seems to work, however NetworkManager is then unable to obtain a IP address via DHCP. The kernel prints the following error on each try:

```
ieee80211 phy0: brcmf_escan_timeout: timer expired
```

The kernel used is the same version as Raspberry Pi OS is using currently (built from tag `1.20221104`). 

The OS uses a fairly standard user space using wpa_supplicant v2.10 and NetworkManager v1.34.0.

I am using a original Raspberry Pi power supply (with 5.1V).

On the same hardware Raspberry Pi OS seems to work (using NetworkManager as well). So this must be some kind of user space configuration or race condition.

Is there a specific wpa_supplicant or NetworkManager patch/configuration required? I understand that this issue is probably beyond what Raspberry Pi supports. But maybe this issue has been seen previously and/or you have other insights which might help me get this problem resolved.

See also: https://github.com/home-assistant/operating-system/issues/2220

pelwell, 2022-12-29:
Just a quick note to say that I've seen this, but will be unable to investigate until the new year. No obvious causes spring to mind, but you might be able to get some insight by enabling the output of messages from the firmware to the kernel log. Add `brcmfmac.debug=0x100000` to `/boot/cmdline.txt`, keeping everything in a single line, and reboot. After the DHCP failure, have a look in the output of `dmesg` for any clues - `dmesg | grep brcmfmac` should help to reduce the clutter.

TiceRex, 2023-01-03:
Thank you for your help in advance!
Attached please find the RPi Zero 2W logs with debug enabled.

dmesg (full): [dmesg.txt](https://github.com/RPi-Distro/firmware-nonfree/files/10339359/dmesg.txt)
dmesg (grep -i brcm): [dmesg_brcm.txt](https://github.com/RPi-Distro/firmware-nonfree/files/10339360/dmesg_brcm.txt)
journalctl -b 0 -u NetworkManager: [journal_NetworkManager.txt](https://github.com/RPi-Distro/firmware-nonfree/files/10339361/journal_NetworkManager.txt)

TiceRex, 2023-01-30:
Is there anything else to try in order to investigate the cause?
Does the debug lines help in any way?
```
[  126.677296] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[  127.287290] brcmfmac: CONSOLE: wl0: wlc_enable_probe_req: state down, deferring setting of host flags
[  127.927288] brcmfmac: CONSOLE: wl0: link up (wl0)
[  127.927317] brcmfmac: CONSOLE: wl0.0: wlc_send_bar: seq 0x1 tid 0
[  127.927326] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  127.927334] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  129.457222] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  173.097324] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  173.097355] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  173.097364] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  173.637291] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  173.657282] brcmfmac: CONSOLE: wl0: wlc_iovar_op: BCME -23 :ndoe
[  183.516904] ieee80211 phy0: brcmf_escan_timeout: timer expired
```