RPi-Distro/firmware-nonfree #33: Raspberry Pi Zero 2 W - Stuttering sound over Bluetooth

cipriantrofin, 2022-12-25:
I suspect there is an interference problem on Raspberry Pi Zero 2 W when using WiFi and Bluetooth.
I'm getting choppy / stuttering sounds when playing MP3s from my NAS and sending the output to an external Bluetooth speaker.

To replicate, I did the following:
1. fresh Raspberry Pi OS Lite 64bit install (using the official Raspberry Pi Imager)
2. `sudo apt update && sudo apt upgrade -y` + reboot
3. add pulseaudio package - `sudo apt install pulseaudio-module-bluetooth` + reboot
4. connect to external BT speaker using bluetoothctl (scan for spearker / connect to speaker)
5. install mpg123 - `sudo apt install mpg123`
6. mount a Samba share from my NAS and play an MP3 from the share, using mpg123; I get significant stutter.

pelwell, 2022-12-29:
1. WiFi and Bluetooth operate in the same frequency bands, but it's worse for some channels than others - try changing your AP channel to see if it improves matters.

2. Do you have experience with the older Pi Zero W and, if so, is Zero 2 W worse in this regard? The sharing mechanism between Bluetooth vs WiFi was adjusted in a CYW43438 firmware (as used on Zero W). Zero 2 W requires a different firmware, and it's possible that some similar tuning could help, but there are always going to be limitations when they share the same antenna.

cipriantrofin, 2022-12-31:
Phil, I don't have direct experience with older Pi Zero W.

With Pi Zero 2 W I had similar stuttering / choppy sound issues with a couple of firmware versions:
`Firmware: BCM43430/**1** wl0: Feb  5 2021 12:49:07 version 7.45.96 (r745790) FWID 01-71817851 es7` and
`Firmware: BCM43430/**1** wl0: Jul  5 2022 21:29:56 version 7.45.96.141 (g53ba0094) FWID 01-eef9b3bf es7`

I came across this issue using the Moode Audio aplication (better described here: https://moodeaudio.org/forum/showthread.php?tid=4937) and - following the suggestion of Moode forum users, I decided to dwelve into it.
I know forum reports are not first hand experience, but could be still relevant:

- NO interference when using Raspberry Pi Zero W v1.1, Pi3A and Pi3B+
- similar interference with Pi Zero 2 W, with BCM43430/**2** forum.

I had no improvement using 1, 6 and 11 WiFi channels.

pelwell, 2023-01-10:
The relevant changes are actually to the "nvram" files - text files containing various settings - which means it should be easy to test a fix.

Note that in my limited testing now I've not seen a problem with stuttering, but the WiFi performance was suspiciously lousy so it isn't really a fair test. This means that I have no real confidence that these changes will improve things, but it's a simple change and easy to revert so worth a try.

I'm assuming from your comment above that your Zero 2 W is the 43430/1 variant, but the change is the same for both, just to different files:
* If `dmesg | grep brcmfmac` contains `using brcm/brcmfmac43436s-sdio for chip BCM43430/1`, edit `/lib/firmware/brcm/brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt`.
* If `dmesg | grep brcmfmac` contains `using brcm/brcmfmac43436-sdio for chip BCM43430/2`, edit `/lib/firmware/brcm/brcmfmac43436-sdio.raspberrypi,model-zero-2-w.txt`.

The proposed change, which I think will be the same for both devices, is to append the following lines:
```
# Improved Bluetooth coexistence parameters
btc_mode=1
btc_params8=0x4e20
btc_params1=0x7530
```
Obviously, reversion is simply a matter of deleting them.

Give it a try and let me know how you get on.

pelwell, 2023-01-10:
Update: with a different test setup I do see stuttering without the change and an improvement with it, but the improvement comes at the cost of a large reduction in throughput while there is Bluetooth traffic - unfortunately that is always going to be the case.

cipriantrofin, 2023-01-10:
Thank you, @pelwell. I followed your instructions and it looks solved.

I don't know how to check the reduction in WiFi throughput, but - in my use case - there is enough left to stream Internet Radio or mp3s from local network,

closed 2023-01-10 by cipriantrofin

pelwell, 2023-01-11:
Thanks for the feedback. These coexistence parameters will be present in future releases.

sudo-splinter-cell, 2024-03-24:
Are we absolutely certain that these lines are necessary accross the board with all Pi devices out there? I was searching for a way to fix my bluetooth audio issues on my Raspberry Pi-5 since the day i purchased it. Same kind of issues were also present with my Pi-4 (but less frequent). Pulsaudio or pipewire, all the same. Disabled wifi, still had issues. I then came up with the forum post that directed me to here. I checked my "relevant" file (for RPI-5 it is brcmfmac43455-sdio.txt). I found that the lines mentioned in this bug report (and one more line) was added to this file also, even though it is a different kind of chip on a different kind of Pi. The lines were:

```
#btc_mode=1
#btc_params8=0x4e20
#btc_params1=0x7530
#btc_params50=0x972c
```

Honestly, i have no idea what these lines do, but as you can see, i have commented them out, just to see if it will change things for the better for me. It is too early to say but i am having very fewer and far in between bluetooth audio stutters or hiccups since i have commented them out. I will continue testing for a week and share my results.

PS: I am on a 5ghz wifi connection. Haven't tried 2.4 ghz. But i assume these lines were added to make it possible for 2.4ghz wifi and bluetooth to coexist. But maybe, to fix this issue, they somewhat "hamper" the bluetooth audio connections. I think this deserves some investigation.

nicolas-chaulet, 2024-05-27:
I just came accroos this, I ahd a cracking sounds that was not going away. The `brcmfmac43436s-sdio.raspberrypi,model-zero-2-w.txt` had the following lines
```
# Improved Bluetooth coexistence parameters
btc_mode=1
btc_params8=0x4e20
btc_params1=0x7530
```
Commenting them out fixed the issue! Looks like those are needed for older devices maybe?

DanoBoca, 2024-07-21:
I've been pulling my hair out trying to solve bluetooth stuttering on a PI 5. This solution works!  Thank  you.

joopdo, 2025-01-30:
Not to revive an old issue, but I found this discussion helpful in debugging my Raspberry 2W. I use it to stream audio via Wifi, but also play Bluetooth from my phone. 
Nothing seemed to help, until I tweaked as follows, now both work:
```
btc_mode=5
btc_params8=5000
btc_params9=40000
btc_params50=0x2000
```

Bluetooth is not stuttering and wifi seems solid too:
```
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   896 KBytes  7.33 Mbits/sec                  
[  5]   1.00-2.00   sec  1.38 MBytes  11.5 Mbits/sec                  
[  5]   2.00-3.00   sec  1.75 MBytes  14.7 Mbits/sec                  
[  5]   3.00-4.00   sec  1.75 MBytes  14.7 Mbits/sec                  
[  5]   4.00-5.00   sec  1.12 MBytes  9.44 Mbits/sec                  
[  5]   5.00-6.00   sec  1.25 MBytes  10.5 Mbits/sec                  
[  5]   6.00-7.00   sec  1.50 MBytes  12.6 Mbits/sec                  
[  5]   7.00-8.00   sec  1.38 MBytes  11.5 Mbits/sec                  
[  5]   8.00-9.00   sec  1.50 MBytes  12.6 Mbits/sec                  
[  5]   9.00-10.00  sec  1.50 MBytes  12.6 Mbits/sec       
```