RPi-Distro/firmware-nonfree #35: Compute module 4 symlink confusion

BubuOT, 2023-01-24:
3b108c864428dda9f9833cf3346f438552d08cea added symlinks for the pi models. I don't exactly know when and why these are used but I got confused by the `brcmfmac43456-sdio.raspberrypi,4-compute-module.bin` existing for the `brcmfmac43456` and not the `brcmfmac43455` / `cyfmac43455`. The [CM4 data sheet](https://datasheets.raspberrypi.com/cm4/cm4-datasheet.pdf) mentions that it comes with the `Cypress CYW43455` and the kernel seems to indeed try and load the `brcm/brcmfmac43455-sdio` for when run on a CM4. Are the symlinks wrong? Am I misunderstanding their usage? Or are there different CM4 reversions and this symlink is only used for ones that do actually have `brcmfmac43456`?

pelwell, 2023-01-25:
The CM4's I have - one that was on the desk, and one I just pulled out of stores - are equipped with 43456s, so those symlinks aren't wrong. Given the chip shortages we've been experiencing, I'm not exactly surprised that there are also some built with 43455s - the two parts are very similar, but with different suppliers and firmwares.

As you've guessed, the kernel first establishes which part is fitted, then tries to load the firmware and associated files, first using a name qualified with the board name, and then without. All that adding these symbolic links has done is remove some warning messages from the kernel log about failed attempts to loading the files with the qualified names. A 43455-equipped CM4 hasn't crossed my path yet, but I'm happy to add the equivalent symlinks.

See https://github.com/RPi-Distro/firmware-nonfree/commit/7f29411baead874b859eda53efdc2472345ea454, which will appear in future releases.

BubuOT, 2023-01-25:
Thanks for your quick reply, that clears up my confusion. I haven't actually seen the kernel warnings, even though we apparently didn't have the CM4 43455 symlinks... but there were Pi4B symlinks in place, maybe those got used as a fallback instead. In any case, I don't actually care about those warnings, as long as we have the correct firmware. :-)

closed 2023-01-25 by BubuOT