RPi-Distro/firmware-nonfree #36: "Unnamed/non-netdev P2P-device" purpose ?

gdoffe, 2023-04-12:
Hello everyone,

The generic behavior I describe further was reproduced on Pi4 and CM3+ using various contexts: custom Yocto Dunfell Linux distribution or Raspberry Pi OS with linux-rpi kernel or mainline kernel.
Both Pi4 and CM3+ are using a BCM43455 WiFi/BLE combo chip.

My need is the following: having the AP + Managed modes working at the same time. But if the Managed interface change its frequency due to new connection, the AP frequency must follow dynamically Managed one.

To achieve this I use  `cypress/cyfmac43455-sdio.bin` firmware from the mainline `linux-firmware` tag [20211216](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tag/?h=20211216).

I create the `wlan0_ap` device using the following command:
```
/usr/sbin/iw dev wlan0 interface add wlan0_ap type __ap
```
It gives me the following interfaces:
```
root@revpi-cm3:~# iw dev
phy#0
	Interface wlan0_ap
		ifindex 5
		wdev 0x2
		addr c4:ac:59:79:85:d9
		ssid TESTING_AP
		type AP
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 31.00 dBm
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr c4:ac:59:79:85:d9
		type managed
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 31.00 dBm
```
In this configuration on my CM3, if I want to change the `wlan0` connection for an other AP with an other channel, `wlan0_ap` channel does not follow the new `wlan0` frequency.

I noticed on my Pi4 with a fresh install that there is a `P2P-device` interface that seems marked as `Unnamed/non-netdev`.
If I manually create this interface on my previous setup on CM3 (with this command), the magic happens and every time the managed interface connects to an other external AP, the wlan0_ap` interface channels follow the `wlan0` one.
```
/usr/sbin/iw dev wlan0 interface add wlan0_p2pdev type __p2pdev
```
```
root@revpi-cm3:~# iw dev
phy#0
	Unnamed/non-netdev interface
		wdev 0x3
		addr c6:ac:59:79:85:d9
		type P2P-device
	Interface wlan0_ap
		ifindex 5
		wdev 0x2
		addr c4:ac:59:79:85:d9
		ssid TESTING_AP
		type AP
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 31.00 dBm
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr c4:ac:59:79:85:d9
		type managed
		channel 5 (2432 MHz), width: 20 MHz, center1: 2432 MHz
		txpower 31.00 dBm
```

So I have the expected behavior, but what is the purpose of this `Unnamed/non-netdev` `P2P-device` interface please ?

pelwell, 2023-04-12:
The "Issues" section is meant for problem reports, not random "I'm curious" questions. I can't give you an answer, other than the obvious "something to do with peer-to-peer networking". Presumably its useful to have a dedicated interface for it (separate IP addresses and controllability, etc.?), just as the AP and client interfaces are separate even though there is just one wireless chip.

closed 2023-04-12 by pelwell

gdoffe, 2023-04-12:
@pelwell thank you so much for this very useful answer.

You got me right, I'm actually curious by nature. Just like the company I work for and who asks me why it is necessary to have a P2P-Device interface, when their 25 000 products based on a CM3+ just need a Managed and AP interfaces. :thinking: 

It becomes an issue/problem when you discover a such unusual behavior. For now, you closed the issue (thanks for that), without really answering it. We do not build products on doubts.
Non-answer let me think you do not know why this P2P-Device is needed. I'm fine with that, but just point me where I could find some help. If my understanding of this problem is not firmware related, point me on the possible source of the problem.

For now, the firmware from this repo is creating the `Unnamed/non-netdev interface` `P2P-Device` by default, the mainline linux-firmware does not. Why ?

ghollingworth, 2023-04-12:
This looks useful...  Maybe you should start here:

https://wireless.wiki.kernel.org/en/users/documentation/iw

pelwell, 2023-04-12:
> For now, the firmware from this repo is creating the Unnamed/non-netdev interface P2P-Device by default, the mainline linux-firmware does not. Why ?

I think you are mistaken. Earlier you said:
> I noticed on my Pi4 with a fresh install that there is a P2P-device [...]

This is where difference lies, not in the wireless firmware. Raspberry Pi OS uses wpa_supplicant, and it is requesting the creation of the P2P device interface.

Compare our firmware:
```
pi@raspberrypi:~$ dmesg | grep "Firmware: BCM4345"
[    8.940717] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
pi@raspberrypi:~$ iw dev
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr de:a6:32:91:36:7c
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr dc:a6:32:91:36:7c
                type managed
                channel 116 (5580 MHz), width: 20 MHz, center1: 5580 MHz
                txpower 31.00 dBm
```
with upstream firmware:
```
pi@raspberrypi:~$ dmesg | grep "Firmware: BCM4345"
[    9.367518] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
pi@raspberrypi:~$ iw dev
phy#0
        Unnamed/non-netdev interface
                wdev 0x2
                addr de:a6:32:91:36:7c
                type P2P-device
                txpower 31.00 dBm
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr dc:a6:32:91:36:7c
                type managed
                channel 36 (5180 MHz), width: 20 MHz, center1: 5180 MHz
                txpower 31.00 dBm
```
Your Yocto build must be using a different configuration that doesn't request the P2P device, either because it doesn't use wpa_supplicant or because its wpa_supplicant is configured differently.