RPi-Distro/firmware-nonfree #38: CTRL-EVENT-ASSOC-REJECT status_code=16

hmeerlo, 2023-09-29:
Heads-up: this is a cross post from  the`raspberrypi/firmware` repo because I realized it belonged here.

The internet is filled with this issue. We have been trying to find and fix this problem for weeks now. But there is no solution yet. We have a consumer product using the cm4. It has the cypress/infineon 43455 chip and the problem we're facing appears to come from the firmware on this chip. Some customers have this problem, others do not experience it (so it seems to depend on the wifi environment as well).
Every now and then the system log starts spewing these messages:

```
2023-09-24 17:40:33.1628 wpa_supplicant[299]: nl80211: Ignored event 46 (NL80211_CMD_CONNECT) for foreign interface (ifindex 3 wdev 0x0)
2023-09-24 17:40:33.1631 wpa_supplicant[299]: nl80211: Drv Event 46 (NL80211_CMD_CONNECT) received for wlan0
2023-09-24 17:40:33.1634 wpa_supplicant[299]: nl80211: Connect event (status=16 ignore_next_local_disconnect=0)
2023-09-24 17:40:33.1637 wpa_supplicant[299]: wlan0: Event ASSOC_REJECT (12) received
2023-09-24 17:40:33.1640 wpa_supplicant[299]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
2023-09-24 17:40:33.1643 wpa_supplicant[299]: wlan0: Radio work 'connect'@0x559d6244f0 done in 0.901671 seconds
```

It then fails to connect to the AP. We would like to know what this `status=16` means. Maybe when we know this we can try to work around it. It is not easily reproducable in a lab environment unfortunately.

I attached a full log of the problem where it sometimes connects but also gets this error quite often.
[1ED95DB6-github.txt](https://github.com/raspberrypi/firmware/files/12748901/1ED95DB6-github.txt)

**System**

* Which model of Raspberry Pi? CM4
* Which OS and version (`cat /etc/rpi-issue`)? 
```
Raspberry Pi reference 2023-09-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7073d28f22d93220d610ecaf7106fe87914746d6
```
* Which firmware version (`vcgencmd version`)?
```
Mar 17 2023 10:50:56
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start_cd)
```
* Which kernel version (`uname -a`)? `Linux hostname 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

hmeerlo, 2023-10-16:
Is there anybody who can comment on this issue? The wifi experience on the cm4 is really really bad. We have so many customers complaining with all kinds of issues. So the above is the not the only problem.

pelwell, 2023-10-16:
The problem with many of these reported WiFi problems is that they lack sufficient information to reproduce them - as you say, it depends on the environment, and clearly our environments don't tickle the problem.

One thing that might be worth a try is disabling the firmware's roaming capability. Does adding `brcmfmac.roamoff=1` to `cmdline.txt` make a difference?

hmeerlo, 2023-10-16:
Oh I have already done that. Doesn't make any difference. We have thousands of customers, so they have many diverse wifi environments. I personally also don't have the problem, so it does seem to be environment related. But one of my colleagues has the problem on a daily basis. It manifests in a different way, but it mostly comes down to 4way handshake problems. And I think they almost always originate in the wifi driver. We are already testing the newly released infineon driver but that doesn't improve the situation. This is one of the other failures:

```
2023-10-13 13:27:06.1723 wpa_supplicant[301]: RTM_NEWLINK: ifi_index=3 ifname=wlan0 wext ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
2023-10-13 13:27:06.1726 wpa_supplicant[301]: nl80211: Event message available
2023-10-13 13:27:06.1728 wpa_supplicant[301]: nl80211: Ignored event 48 (NL80211_CMD_DISCONNECT) for foreign interface (ifindex 3 wdev 0x0)
2023-10-13 13:27:06.1729 wpa_supplicant[301]: nl80211: Drv Event 48 (NL80211_CMD_DISCONNECT) received for wlan0
2023-10-13 13:27:06.1731 wpa_supplicant[301]: nl80211: Disconnect event
2023-10-13 13:27:06.1732 wpa_supplicant[301]: wlan0: Event DEAUTH (11) received
2023-10-13 13:27:06.1734 wpa_supplicant[301]: wlan0: Deauthentication notification
2023-10-13 13:27:06.1735 wpa_supplicant[301]: wlan0:  * reason 2 (PREV_AUTH_NOT_VALID)
2023-10-13 13:27:06.1736 wpa_supplicant[301]: Deauthentication frame IE(s) - hexdump(len=0): [NULL]
2023-10-13 13:27:06.1738 wpa_supplicant[301]: wlan0: CTRL-EVENT-DISCONNECTED bssid=48:d3:43:92:c9:5f reason=2
2023-10-13 13:27:06.1739 wpa_supplicant[301]: wlan0: Auto connect enabled: try to reconnect (wps=0/0 wpa_state=9)
2023-10-13 13:27:06.1740 wpa_supplicant[301]: wlan0: Setting scan request: 0.100000 sec
2023-10-13 13:27:06.1742 wpa_supplicant[301]: Added BSSID 48:d3:43:92:c9:5f into blacklist
```

This apparently means that the client (cm4) did not respond to one of the first messages of the handshake.

hmeerlo, 2023-11-07:
This sums the situation up pretty well: https://rachelbythebay.com/w/2023/11/06/wpa3/ How can you start selling a brand new product with a driver that is so extremely outdated and bad? Raspberry Pi really needs to make a proper plan for this and get Infineon on board in updating this driver to the latest standards.

pimlie, 2023-11-23:
> The problem with many of these reported WiFi problems is that they lack sufficient information to reproduce them - as you say, it depends on the environment, and clearly our environments don't tickle the problem.

@pelwell Using a RPi 0W2 with the latest 64bit image (tried both desktop & lite) I can consistently reproduce this error trying to connect to 3 different Mikrotik devices running two different software versions (routeros latest v7 using wifiwave2 & latest v6). The 0W2 connects fine to a hotspot on my phone, but I just cannot connect to any of those 3 Mikrotik ap's. The only thing the mikrotik devices report is that `unicast key exchange failed`. Does that help to try to narrow it down? Let me know if you need more info.

- Using the same sd-card in a RPi 4 connects to the wifi without issue.
- With the RPi 4 I could run `apt update && apt upgrade`, this installed some newer firmware. After rebooting, wifi would still connect using the RPi 4, but
- after inserting the sd-card (ie with latest updates) back into the 0W2 the wifi still fails to connect.
- Unfortunately I only have one 0W2 to test with

parsnatura, 2024-02-29:
@pimlie Thx, it helped with my RPi 0W2, too. Btw also using Mikrotik RouterOS v7.

luftdieb, 2024-04-06:
Hi together, I had same error code 18 with my Dell Latitude 6520 Laptop and Mikrotik router RB4011iGS / Firmware v7.14.1 configured with CAPsMAN. 
In Channels configuration I had previously selcted Band "5Ghz-onlyac". But after reading this post, I''ve got the idea changing to "5Ghz-n/ac" and now, Dell Laptop is connecting fine and no error 18 is reported anymore. 

Beside this, I have this additional settings, which I'm not sure, if it makes sense or not... but it works fine on my end now.
Freqzency: 5180, 5260,5500,5580,5660,5745
Control Chanel With: 20 Mhz
Band: "5Ghz-n/ac"

Maybe this helps others :-)

Zanz2, 2024-04-22:
> > The problem with many of these reported WiFi problems is that they lack sufficient information to reproduce them - as you say, it depends on the environment, and clearly our environments don't tickle the problem.
> 
> @pelwell Using a RPi 0W2 with the latest 64bit image (tried both desktop & lite) I can consistently reproduce this error trying to connect to 3 different Mikrotik devices running two different software versions (routeros latest v7 using wifiwave2 & latest v6). The 0W2 connects fine to a hotspot on my phone, but I just cannot connect to any of those 3 Mikrotik ap's. The only thing the mikrotik devices report is that `unicast key exchange failed`. Does that help to try to narrow it down? Let me know if you need more info.
> 
> * Using the same sd-card in a RPi 4 connects to the wifi without issue.
> * With the RPi 4 I could run `apt update && apt upgrade`, this installed some newer firmware. After rebooting, wifi would still connect using the RPi 4, but
> * after inserting the sd-card (ie with latest updates) back into the 0W2 the wifi still fails to connect.
> * Unfortunately I only have one 0W2 to test with
> 
> After much more searching I came accross [this SO](https://raspberrypi.stackexchange.com/questions/77144/rpi3-wireless-issue-ctrl-event-assoc-reject-status-code-16), and indeed turning off fast roaming & disabling SAE for the brcmfmac module made the 0W2 finally connect succesfully
> 
> ```
> modprobe brcmfmac roamoff=1 feature_disable=0x82000
> ```

Same pi, same hardware, the mikrotik worked on both my pi3's also my pi5, but not my zeros, your fix worked, i sunk 2 whole days into this. If i could i'd buy you a coffee

txoof, 2024-08-06:
I have a similar issue, but it seems to only impact raspberry pi 3 units. Both units have static IPs, both I've checked the router and the DHCP lease is set for one day.

I have two Pi3 and two Pi4 running on the local network and the problem always occurs at 38 minutes past the hour and only impacts the Pi3 units, and always with `reason=16`.

Restarting the NetworkManager service temporarily resolves this issue. Wired connections are not impacted.

The Pi4 and Pi3 are configured using the same minimal setup. All four devices are on the same release:
```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
Here's a sample of the wpa_supplicant errors occurring at 38 minutes past the hour:
```
ug 03 15:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 15:38:56 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=6
Aug 03 17:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 17:38:06 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=15
Aug 03 18:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 19:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 20:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 21:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 22:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 03 23:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 00:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 02:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 03:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 04:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 04:38:07 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=15
Aug 04 05:38:01 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 05:38:06 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=15
Aug 04 07:38:02 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 08:38:02 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 08:38:06 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=15
Aug 04 09:38:02 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 10:38:02 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 04 12:38:02 office-paperpi wpa_supplicant[563]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 05 09:38:02 office-paperpi wpa_supplicant[388859]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 05 10:38:02 office-paperpi wpa_supplicant[388859]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 05 11:38:02 office-paperpi wpa_supplicant[388859]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16
Aug 05 11:38:07 office-paperpi wpa_supplicant[388859]: wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=15
```
Here's  the full error output:
```
2024-08-05T12:38:02.334 2633844492 wpa_supplicant.service wpa_supplicant[571] STATS wlan0: CTRL-EVENT-DISCONNECTED bssid=e0:63:da:18:43:db reason=16                                                                                │
│2024-08-05T12:38:02.343 2633853589 wpa_supplicant.service wpa_supplicant[571] STATS wlan0: CTRL-EVENT-REGDOM-CHANGE init=CORE type=WORLD                                                                                            │
│2024-08-05T12:38:02.345 2633855621 wpa_supplicant.service wpa_supplicant[571] STATS wlan0: CTRL-EVENT-REGDOM-CHANGE init=USER type=COUNTRY alpha2=NL                                                                                │
│2024-08-05T12:38:02.350 2633860104 NetworkManager.service NetworkManager[568] INFO <info>  [1722854282.3494] device (wlan0): supplicant interface state: completed -> disconnected                                                  │
│2024-08-05T12:38:02.351 2633862032 NetworkManager.service NetworkManager[568] INFO <info>  [1722854282.3495] device (p2p-dev-wlan0): supplicant management interface state: completed -> disconnected                               │
│2024-08-05T12:38:02.442 2633952728 NetworkManager.service NetworkManager[568] INFO <info>  [1722854282.4424] device (wlan0): supplicant interface state: disconnected -> scanning                                        [... cut ...]
│2024-08-05T12:38:03.183 2634693265 NetworkManager.service NetworkManager[568] INFO <info>  [1722854283.1823] dhcp4 (wlan0): state changed no lease                                                                                  │
│2024-08-05T12:38:03.183 2634693570 NetworkManager.service NetworkManager[568] INFO <info>  [1722854283.1826] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)                                               │
│2024-08-05T12:38:03.400 2634911011 NetworkManager.service NetworkManager[568] INFO <info>  [1722854283.4004] dhcp4 (wlan0): state changed new lease, address=192.168.1.127
```

radu-v, 2024-09-04:
> One thing that might be worth a try is disabling the firmware's roaming capability. Does adding `brcmfmac.roamoff=1` to `cmdline.txt` make a difference?

this worked for me, on Raspberry Pi 5, thanks @pelwell!

irabbit79, 2024-11-20:
Had the same problem when connecting USB3 HDD , https://forums.raspberrypi.com/viewtopic.php?t=328248

txoof, 2024-11-24:
I eventually found that the problem stemmed from the _Group Rekey Interval_ on the APs. The reason the the problem reappeared every hour was related to the rekey interval being 3600 seconds. I changed the rekey interval to 1800 seconds and the problem disappeared. After a few weeks of stability, I moved the rekey interval back to 3600 and the problem has not returned. 

I honestly have no idea why this would have anything to do with anything, but it solved my problem.

> I have a similar issue, but it seems to only impact raspberry pi 3 units. Both units have static IPs, both I've checked the router and the DHCP lease is set for one day.
>
> I have two Pi3 and two Pi4 [...]

pelwell, 2024-11-24:
Thanks - that's a useful observation.

marcone, 2024-11-25:
> Had the same problem when connecting USB3 HDD , https://forums.raspberrypi.com/viewtopic.php?t=328248

Same here with a Raspberry Pi 4.
Curiously, I had originally connected the USB drive to the top USB3 port, and even tried an extension cable to get the drive away from the Pi, but wifi just wouldn't connect even after numerous retries/reboots (the AP is a 2.4 GHz GL-MT300A travel router).
After connecting the USB drive to the bottom USB3 port, wifi started working, and still works after plugging the drive back in to the top port.

juju4, 2025-01-12:
same issue here on a Pi3, Raspbian 12 (6.6.62-1+rpt1 (2024-11-25) armv7l). Curiously happened after more than a month of stable wifi usage and after keep losing wifi every couple of days if not more.

`modprobe brcmfmac roamoff=1 feature_disable=0x82000`  + rc.local/reboot
= still happened a day after.

```
kernel: [ N.N] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
kernel: [ N.N] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
kernel: [ N.N] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob available (err=-2)
kernel: [ N.N] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Jul 19 2021 03:24:18 version 7.45.98 (TOB) (56df937 CY) FWID 01-8e14b897
kernel: [ N.N] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

Note than logs "CTRL-EVENT-ASSOC-REJECT status_code=16" appeared once in a while before but without impact, while, since Jan 8, it's more a dozen times a day for me.

defanator, 2025-03-19:
Found this issue from multiple search queries. I've been running Raspberry Pi 5 headless for a few weeks, and recently connected it to portable monitor via HDMI cable (Ugreen converter from micro-HDMI to HDMI + some regular cable) + plugged in radio USB keyboard. Shortly after this my RPi lost wifi connection (I'm using 2.4Ghz network with TPLink Archer C7 router). I spent some time trying to understand what has caused the issue, and surprisingly it turned out to be HDMI connection.

If I remove HDMI cable from RPi, it can connect to my 2.4Ghz wifi network just fine (with or without radio keyboard dongle).

If I plug in HDMI cable, wifi stops working. I also tried 5Ghz AP (on the same router) - got exactly the same behavior with slightly different errors in wpa_supplicant logs.

wpa_supplicant logs (with monitor connected):
```
Mar 19 14:51:51 rpi5-mon wpa_supplicant[640]: nl80211: kernel reports: Registration to specific type not supported
Mar 19 14:51:58 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:51:58 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:52:05 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:52:05 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:52:12 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:52:13 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:52:20 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:52:21 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:52:21 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MYSSID" auth_failures=1 duration=10 reason=CONN_FAILED
Mar 19 14:52:23 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:52:23 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:52:23 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MYSSID" auth_failures=1 duration=10 reason=CONN_FAILED
Mar 19 14:52:39 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="MYSSID"
Mar 19 14:52:51 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:52:52 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:52:52 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MYSSID" auth_failures=1 duration=10 reason=CONN_FAILED
Mar 19 14:53:08 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="MYSSID"
Mar 19 14:53:08 rpi5-mon wpa_supplicant[640]: wlan0: Trying to associate with SSID 'MYSSID'
Mar 19 14:53:09 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
Mar 19 14:53:09 rpi5-mon wpa_supplicant[640]: BSSID c0:c9:e3:3b:7e:18 ignore list count incremented to 2, ignoring for 10 seconds
Mar 19 14:53:09 rpi5-mon wpa_supplicant[640]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="MYSSID" auth_failures=2 duration=20 reason=CONN_FAILED
```

I find this quite fascinating.

ghollingworth, 2025-03-20:
Sounds like either the HDMI cable or the HDMI monitor does not pass the CE/FCC regulations for radiated emissions...  Are you using an official Raspberry Pi HDMI cable, or something you got from Amazon?

chadaustin, 2025-05-18:
I have a Pi Zero unable to connect to one of my access points after upgrading to Bookworm. I see the same CTRL-EVENT-ASSOC-REJECT status_code=16. I tried all of the workarounds listed: different wifi channels, disabling WMM on the AP, disabling scan MAC randomization, unplugging HDMI. Nothing helped.

The AP is running OpenWRT. I wondered if it was a Network Manager vs. wpa_supplicant issue, so I downgraded to Bullseye and wifi works reliably again.

I have yet to see whether this issue persists on a Pi Zero 2.

amisix, 2025-05-21:
Raspberry PI CM4 / Onboard Broadcomm wifi adapter
OpenWRT 24.10.1 

Attempting to connect Raspberry Pi CM4 to AP as a station - will not connect. 
Able to connect to AP with other devices, just not this Raspberry Pi CM4.
AP is 2.4GHZ / open, no WPA
No HDMI connected/in use on Raspberry Pi CM4

Error: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16

Complete error: 
daemon.notice wpa_supplicant[856]: phy1-sta0: Trying to associate with 90:3a:e6:38:7d:ed (SSID='Bebop2-055156' freq=2442 MHz)
daemon.notice wpa_supplicant[856]: phy1-sta0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
daemon.notice wpa_supplicant[856]: phy1-sta0: Added BSSID 90:3a:e6:38:7d:ed into ignore list, ignoring for 10 seconds
daemon.notice wpa_supplicant[856]: phy1-sta0: CTRL-EVENT-SSID-TEMP-DISABLED id=4 ssid="Bebop2-055156" auth_failures=16 duration=120 reason=CONN_FAILED

It kinda worked for a few days then stopped working entirely and now I'm unable to connect to the AP no matter what trounleshooting I do. 

Troubleshooting: 
Configured AP with no WPA key (open) - failed
Configured AP with WPA2 key and tried connecting with Raspberry Pi - failed
Tried connecting via BSSID instead of ESSID - failed
Tried changing AP channel - no change

RyoheiHashimoto, 2026-09-08:
## What `status_code=16` actually is (driver-level trace)

The original question in this issue — *"We would like to know what this `status=16` means"* —
can be answered from the driver's own debug output. **It is not an AP response.** The firmware
rejects the join request locally and the driver synthesises the 802.11 status code.

### Short answer: `16` is a hardcoded placeholder, not information

`brcmfmac` can only ever report two connect statuses. From
`drivers/net/wireless/broadcom/brcm80211/brcmfmac/cfg80211.c`,
`brcmf_bss_connect_done()`:

```c
memset(&conn_params, 0, sizeof(conn_params));
if (completed) {
        ...
        conn_params.status = WLAN_STATUS_SUCCESS;
} else {
        clear_bit(BRCMF_VIF_STATUS_EAP_SUCCESS, &ifp->vif->sme_state);
        clear_bit(BRCMF_VIF_STATUS_ASSOC_SUCCESS, &ifp->vif->sme_state);
        conn_params.status = WLAN_STATUS_AUTH_TIMEOUT;
}
conn_params.links[0].bssid = profile->bssid;
cfg80211_connect_done(ndev, &conn_params, GFP_KERNEL);
```

`WLAN_STATUS_AUTH_TIMEOUT` is `16` (`include/linux/ieee80211.h`). These are the only two
`WLAN_STATUS_*` values anywhere in the file. **So `status_code=16` does not mean
"authentication timed out" — it is simply the driver's only way of saying "this connect
attempt did not complete".** It carries no diagnostic information about the cause.

The all-zero BSSID has the same origin: `conn_params` is `memset` to zero and
`profile->bssid` is still zero when the station never associated.

So the useful question is not "what does 16 mean" but **"why did the firmware report the
join as not completed"** — and that is visible only with driver debug enabled.

### How to see it

```
# /etc/modprobe.d/brcmfmac-debug.conf
options brcmfmac debug=0x8400      # BRCMF_CONN_VAL | BRCMF_EVENT_VAL
```

### What the trace shows

Failing attempt:

```
brcmfmac: brcmf_fweh_event_worker event SET_SSID (0:0) ifidx 0 bsscfg 0
brcmfmac: brcmf_fweh_event_worker   version 2 flags 0 status 1 reason 0
wpa_supplicant: wlan0: CTRL-EVENT-ASSOC-REJECT bssid=00:00:00:00:00:00 status_code=16
```

Successful attempt, same machine, same AP, same config:

```
brcmfmac: brcmf_fweh_event_worker event SET_SSID (0:0) ifidx 0 bsscfg 0
brcmfmac: brcmf_fweh_event_worker event LINK (16:16) ... status 0
wpa_supplicant: wlan0: CTRL-EVENT-CONNECTED - Connection to ... completed
```

So: the firmware answers the `SET_SSID` (join) command with **status 1** and never emits
`LINK`. The all-zero BSSID in the wpa_supplicant message is the tell — no frame was ever
received from the AP. **No association request goes on air.** Each failure resolves in
**0.36–0.41 s**, far too fast for an over-the-air timeout, and it is remarkably constant.

### Measurements

Raspberry Pi 4 Model B Rev 1.5, `BCM4345/6`, firmware
`wl0: Aug 29 2023 01:47:08 version 7.45.265 (28bca26 CY) FWID 01-b677b91b`
(package `firmware-brcm80211 1:20260519-1~bpo13+1+rpt1`), kernel 6.18.39,
NetworkManager + wpa_supplicant, WPA2-PSK/CCMP, 2.4 GHz ch11, AP at **-26 dBm**.

| boot | `SET_SSID` | `status 1` | `LINK` | outcome |
|---|---|---|---|---|
| A | 1 | 0 | 1 | connected immediately |
| B | 31 | 31 | 0 | 7 minutes of failures, AP visible at -26 dBm throughout |

Time from boot to association, same hardware and AP, measured with
`journalctl -o short-monotonic`: **22 s / 30 s / 37 s / 68 s / 88 s / 326 s / 632 s**.

### Ruled out here (all measured, not assumed)

- **AP / router.** Failures carry an all-zero BSSID, i.e. no AP response exists.
  Other clients (macOS, phone, ESP32) associate with the same AP without trouble.
- **WPA3 / PMF / transition mode.** `brcmfmac.feature_disable=0x82000` is applied and
  *effective* — `wpa_cli get_capability key_mgmt` no longer lists `SAE`. Still fails.
- **`roamoff`.** Already `1` (Raspberry Pi OS default). Confirmed via
  `/sys/module/brcmfmac/parameters/roamoff`.
- **Signal strength.** -26 dBm during the failing streak.
- **Pinning BSSID and band.** Failures continued unchanged *after* the pin was removed.
- **Client power save**, **band restriction to `bg`**, **host CPU load**.
- **Firmware build.** Identical across boots that succeed and boots that fail. The
  board-specific file `brcmfmac43455-sdio.raspberrypi,4-model-b.bin` is byte-identical
  (md5) to `cypress/cyfmac43455-sdio-standard.bin`, so `update-alternatives` between
  `standard` and `minimal` has no effect on what is actually loaded.

### One more observation (offered as data, not as a claim)

While disconnected and retrying, the firmware emits `BRCMF_E_IF` (action ADD, ifidx 0)
at a very regular **10 Hz**:

```
brcmfmac: brcmf_fweh_event_worker event IF (54:54) ifidx 0 bsscfg 0 addr <mac>
brcmfmac: brcmf_fweh_handle_if_event action: 1 ifidx: 0 bsscfgidx: 0 flags: 0 role: 0
brcmfmac: brcmf_fweh_handle_if_event adding wl0 (<mac>)
```

Boot A: the stream ran 186–195 s and the join at 195.19 s succeeded as it stopped.
Boot B: the stream ran 1180–1614 s and all 31 joins inside it failed.
**This may simply be a companion of the disconnected state rather than a cause** —
with only one successful sample here it cannot be separated. Posting it in case it is
meaningful to someone with firmware visibility.

### Practical mitigation (does not fix the cause)

The failure itself costs ~0.4 s; the retry interval costs ~6.9 s, so >90 % of the time to
connect is spent waiting. `wpa_supplicant` also backs off on its own
(`SSID-TEMP-DISABLED duration=10 → 20 → 40`), so retrying naively makes it worse.
Clearing that with `wpa_cli enable_network <id>` before each `reassociate` keeps retries tight.