RPi-Distro/firmware-nonfree #41: BRCM/Cypress 43455 firmware doesn't support WPA3-SAE

herrernst, 2023-11-23:
At https://github.com/raspberrypi/linux/issues/4718, there's a discussion why internal WiFi doesn't support WPA3-SAE encryption (which is the most secure).
Turns out RPi OS ships with firmware `7.45.241 (1a2f2fa CY) CRC: 959ad1c7 Date: Mon 2021-11-01 00:40:29 PDT Ucode Ver: 1043.2164 FWID 01-703fd60` that doesn't support SAE: 
```
$ curl -sL https://github.com/RPi-Distro/firmware-nonfree/raw/bookworm/debian/config/brcm80211/cypress/cyfmac43455-sdio-standard.bin | grep -i sae -c
0
```
But on "official" linux-firmware repo AND in Debian repos there's a slightly older `7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15 03:06:00 PDT Ucode Ver: 1043.2161 FWID 01-996384e2` that announces "sae" support:
```
 $ curl -sL https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/cypress/cyfmac43455-sdio.bin | strings | tail -n 2 | grep -i sae
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv-gtkoe-roamprof-txbf-ve-sae-dpp-sr-okc-bpd Version: 7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15 03:06:00 PDT Ucode Ver: 1043.2161 FWID 01-996384e2
```
Then iw also shows SAE support on RPi OS 12:
```
$ iw phy | grep -i sae
		* [ SAE_OFFLOAD ]: SAE offload support
		* [ SAE_OFFLOAD_AP ]: AP mode SAE authentication offload support
```
Why does RPi's version not support SAE?

sk8board, 2024-01-07:
@herrernst @XECDesign 

To resolve this problem, would it be as easy as submitting a Pull Request with the older WiFi driver that supports WPA3-SAE?

sk8board, 2024-01-08:
There was a pull request from from two years ago for Update CYW43455 to 7.45.234, but they went with 7.45.231. https://github.com/RPi-Distro/firmware-nonfree/pull/19

Here is the location of the CYW43455 firmware in this repository.
[debian/config/brcm80211/cypress](https://github.com/RPi-Distro/firmware-nonfree/tree/bookworm/debian/config/brcm80211/cypress)

XECDesign, 2024-01-08:
I don't think a PR would help. What we ship is decided between infineon and @pelwell.

pelwell, 2024-01-08:
We are likely to switch to the standard Infineon releases in the relatively near future, but @XECDesign is correct in that a PR won't make a difference either way.

sk8board, 2024-01-08:
@pelwell There are many people who would like Raspberry Pi OS to support WPA3-SAE, including 

https://holtmann.dev/enabling-wpa3-on-raspberry-pi/

https://www.youtube.com/watch?v=yUxpm8ucQB8

Do you know if the new release will support WPA3-SAE?

pelwell, 2024-01-08:
> Do you know if the new release will support WPA3-SAE?

That's the intention, but weirdly the latest Cypress release of the 43455 firmware (which I found here -https://community.infineon.com/t5/Wi-Fi-Bluetooth-for-Linux/Cypress-Linux-WiFi-Driver-Release-FMAC-2023-09-01/td-p/492862) doesn't seem to enable it.

The firmware string for the 2023-09-01 7.45.265 release is:
```
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-wfds-mfp-dfsradar-wowlpf-noclminc-clm_min-obss-obssdump-swdiv-gtkoe-roamprof-txbf-ve-extsae-dpp-sr-okc-bpd Version: 7.45.265 (28bca26 CY) CRC: 68bafb8c Date: Tue 2023-08-29 01:51:02 PDT Ucode Ver: 1043.2170 FWID 01-b677b91b
```
whereas for the 7.4.234 release it is:
```
43455c0-roml/43455_sdio-pno-aoe-pktfilter-pktctx-wfds-mfp-dfsradar-wowlpf-idsup-idauth-noclminc-clm_min-obss-obssdump-swdiv-gtkoe-roamprof-txbf-ve-sae-dpp-sr-okc-bpd Version: 7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15 03:06:00 PDT Ucode Ver: 1043.2161 FWID 01-996384e2
```

The former gives nothing:
```
$ iw list | grep SAE
```
And this is even though the firmware string includes `extsae` ,whereas with the upstream, older firmware you get:
```
$ iw list | grep SAE
                * [ SAE_OFFLOAD ]: SAE offload support
                * [ SAE_OFFLOAD_AP ]: AP mode SAE authentication offload support
```
So it looks like the upstream firmware is more suitable.

But it seems like changing the firmware is only half the problem, if getting WPA3 support also requires switching to iwd.

sk8board, 2024-01-08:
@pelwell 
If you do much research about using WPA3-SAE with Raspberry Pi OS, there are plenty of people who are willing to use IWD.

If switching to IWD is not acceptable, then why not upgrade wpa_supplicant to a newer version (2.10) which supports WPA3-SAE?

https://w1.fi/wpa_supplicant/

pelwell, 2024-01-08:
> If you do much research about using WPA3-SAE with Raspberry Pi OS, there are plenty of people who are willing to use IWD.

I didn't say it was unacceptable, but it's definitely a barrier to entry.

> If switching to IWD is not acceptable, then why not upgrade wpa_supplicant to a newer version (2.10) which supports WPA3-SAE?

My Pi 5 seems to already be running 2.10:
```
pi@raspberrypi:~$ wpa_supplicant -v
wpa_supplicant v2.10
Copyright (c) 2003-2022, Jouni Malinen <j@w1.fi> and contributors
```
But I don't think it's being used, since I'm connected to an AP but wpa_supplicant.conf has no mention of it.

XECDesign, 2024-01-08:
> But I don't think it's being used, since I'm connected to an AP but wpa_supplicant.conf has no mention of it.

If you're using NetworkManager, it talks to wpa_supplicant through a socket rather than trying to populate the conf file.

pelwell, 2024-01-08:
I have upgraded the firmware locally, but apart from the iw list output I'm seeing no signs of WPA3 ability.

t0b3, 2024-01-08:
possibly related to https://github.com/Infineon/wpa3-external-supplicant

pelwell, 2024-01-15:
Well, having tested the proposed firmware with an AP configured to only support WPA3, it does indeed appear to connect. Further work may be required to support AP mode, but in the usual Pi-as-client mode it's looking good.

See #42.

sk8board, 2024-01-15:
@pelwell Thank you for your help in working toward WPA3 function in Raspberry Pi OS!

pelwell, 2024-01-19:
FYI, testing the new firmware on a clean RPiOS image has shown that you only gain the WPA3 support when NetManager is configured to use iwd instead of wpa_supplicant. That's not to say that wpa_supplicant cannot ever support WPA3, but the version we are shipping certainly doesn't.

The steps to enable WPA3 are:
1) Install the new firmware (`sudo apt install firmware-brcm80211` when we push the update).
2) Install iwd (`sudo apt install iwd`).
3) Put the following in `/etc/NetworkManager/conf.d/wifi_backend.conf`:
    ```
    [device]
    wifi.backend=iwd
    ```
4) Reboot.

Obviously the aim is to have this work automatically in a new image, but we're not there yet.

HuggeK, 2024-05-24:
RPI OS seems to be using a older version of the brcmfmac firmware for the wifi chip CYW43455 which is used on Raspberry pi 3B+ and newer. 

_12.5 Bookworm 64-bit | Linux: 6.6.28 64-bit_ `dmesg`  returns: 
`brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2` 
Looking at what OP wrote, it has been downgraded to be inline with Debian?

Interestinlgy on Infineons product page for [CYW43455 ](https://www.infineon.com/cms/en/product/wireless-connectivity/airoc-wi-fi-plus-bluetooth-combos/wi-fi-5-802.11ac/cyw43455/#!designsupport) the old fmac software is listed from 2021-12-08. 

At the time of writing the newest version of [Cypress Linux WiFi Driver Release (FMAC)](https://community.infineon.com/t5/Wi-Fi-Bluetooth-for-Linux/Cypress-Linux-WiFi-Driver-Release-FMAC-2023-11-28/td-p/662238). Which was released 2024-01-18! Seems like it is targeted at a 6.19 kernel - we can now bring it over to Bookworm?

The problems seems like a moderator on the release page says:

> CYW43455 is [Not Recommended for New Design(NRND)](https://www.infineon.com/cms/en/product/wireless-connectivity/airoc-wi-fi-plus-bluetooth-combos/?filterValues=~(search~%2743455)&visibleColumnIds=name,orderOnline,opn,productStatusInfo,941,536,1122,1120,1411,1415,980,1413,1417,284_min,284_max,1060,1277,1343,1275) and is on yearly release which is scheduled for the end of Aug 2024.

Which I found odd when you look at their attached link CYW43455 says "active and preferred".

Many have commented on that release the necessary files/support for CYW43455 is still missing from the package, it is past my technical expertise to find it.

Sidenote: While scouraging forums I came across [this](https://community.infineon.com/t5/AIROC-Wi-Fi-MCUs/WPA3-iwd-support/m-p/478999#M733) from a Infineon rep which said that they did not support IWD - ‎Sep 04, 2023.

herrernst, 2024-05-25:
Apparently, in https://github.com/RPi-Distro/firmware-nonfree/pull/42 it was decided and commited to use the older version `7.45.234 (4ca95bb CY) CRC: 212e223d Date: Thu 2021-04-15 03:06:00 PDT Ucode Ver: 1043.2161 FWID 01-996384e2` which is also used in upstream Linux firmware repo.

So closing this, as it's also already in RPi OS 12 repos.

closed 2024-05-25 by herrernst