RPi-Distro/firmware-nonfree #43: "Firmware has halted or crashed", "firmware trap in dongle"

JamesH65, 2024-04-25:
This case came in via Freshdesk ID: 103141. Power supply might be an issue, not sure it could cause this sort of error though.

There is a full log in the FD case, but in precis:

```
[   94.979799] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  217.767997] brcmfmac: brcmf_dump_obss: dump_obss error (-52)
[  226.592810] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  322.435675] ieee80211 phy0: brcmf_fw_crashed: Firmware has halted or crashed
[  325.086175] brcmfmac: brcmf_sdio_bus_rxctl: resumed on timeout
[  325.086637] brcmfmac: brcmf_sdio_checkdied: firmware trap in dongle
[  325.086732] brcmfmac: dongle trap info: type 0x4 @ epc 0x001c43ce
                 cpsr 0x2000019f spsr 0x200001bf sp 0x0025fe50
                 lr   0x00049bd7 pc   0x001c43ce offset 0x25fdf8
                 r0   0x002567c4 r1   0x00000001 r2 0x00010000 r3 0x18001000
                 r4   0x002567c4 r5   0xffffffff r6 0x00000000 r7 0x00000001
[  325.086746] brcmfmac: brcmf_dump_obss: dump_obss error (-110)
[  325.158607] brcmfmac_wcc: brcmf_wcc_detach: executing
[  325.187487] mmc1: card 0001 removed
[  325.301154] mmc1: new high speed SDIO card at address 0001
[  325.301867] brcmfmac: F1 signature read @0x18000000=0x15264345
[  325.306204] brcmfmac: brcmf_fw_alloc_request: using
brcm/brcmfmac43455-sdio for chip BCM4345/6
[  325.494258] brcmfmac_wcc: brcmf_wcc_attach: executing
[  325.498974] brcmfmac: brcmf_c_process_txcap_blob: no txcap_blob
available (err=-2)
[  325.499356] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6
wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID
01-996384e2
[  336.008690] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```

holta, 2025-03-13:
Thank You @JamesH65 !

Nearly identical to yours above, I've posted a couple of nearly identical WiFi firmware crash errors to:

1. https://github.com/iiab/iiab/pull/3965#issuecomment-2719573931
2. https://github.com/iiab/iiab/pull/3965#issuecomment-2719655020

_If you have any testing strategies for us to help isolate the root cause of these crashes (7.45.265 == cyfmac43455-sdio-standard.bin on RPi 4 and RPi 500), we'd be most appreciative!_ :pray:

holta, 2025-04-01:
@JamesH65 are you aware that adding `brcmfmac.roamoff=1 brcmfmac.feature_disable=0x282000` to /boot/firmware/cmdline.txt solves these quite regular `(-110)` firmware crashes?

As mentioned here:

- https://github.com/raspberrypi/linux/issues/6049#issuecomment-2763035887
- PR iiab/iiab#3965

pelwell, 2025-04-01:
A better way to set these options is in a dedicated file under `/etc/modprobe.d`:
```
$ cat /etc/modprobe.d/brcmfmac.conf
options brcmfmac roamoff=1 feature_disable=0x282000
```
(although I believe that if we were to make such settings official we would put it in `/lib/modprobe.d/brcmfmac.conf`)