RPi-Distro/firmware-nonfree #44: Raspberry Pi Zero 2 W: BCM43430/2 Wifi no reconnect

rdlvm, 2024-06-29:
Problem with Raspberry Pi Zero 2 W: BCM43430/2 after 4-5 days working normally with the latest version of Raspbian 64bits lite available today, the wifi disconnects and does not reconnect (the raspberry pi zero 2w continues working normally)

I have disabled Wi-Fi energy saving but it hasn't solved anything.

Some log messages:

`Jun 27 07:19:08 raspberrypi avahi-daemon[398]: Registering new address record for xxxxx on wlan0.*.
Jun 27 07:19:10 raspberrypi avahi-daemon[398]: Withdrawing address record for xxxxx wlan0.
Jun 27 07:19:11 raspberrypi avahi-daemon[398]: Registering new address record for xxxxx on wlan0.*.
Jun 27 07:19:14 raspberrypi avahi-daemon[398]: Withdrawing address record for xxxxx on wlan0.
Jun 27 07:19:15 raspberrypi avahi-daemon[398]: Registering new address record for xxxxx on wlan0.*.
Jun 27 07:19:16 raspberrypi avahi-daemon[398]: Withdrawing address record for xxxxx on wlan0.`

`dmesg | grep "Firmware: BCM43430"
[   11.631775] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/2 wl0: Mar 31 2022 17:24:51 version 9.88.4.77 (g58bc5cc) FWID 01-3b307371
`
[url=https://ibb.co/yX3KX7d][img]https://i.ibb.co/c1mR1Vk/photo-2024-06-20-22-14-42.jpg[/img][/url]

I have done clean installations from 0 with the same problem, both in 32-bit and 64-bit.

I have two Wifi aps at home with the same SSID (now with roaming disabled, thinking that could be the problem)

pelwell, 2024-06-29:
And has the problem reoccurred since disabling roaming?

rdlvm, 2024-06-29:
> And has the problem reoccurred since disabling roaming?

Yes, the above is without active roaming

rdlvm, 2024-07-05:
Any solution for this? Compromise my device to keep working.