RPi-Distro/firmware-nonfree #45: Raspberry Pi 400: BCM43456 firmware does not support 4-way handshake offloading

rps102, 2024-07-05:
This issue is not dissimilar to that described in #23 except that the 43456 (used exclusively on the Pi 400 it seems) doesn't support firmware offload of handshaking (enabled by posting a message "sup_wpa" set to 1 to the firmware).

That's fine if you're Linux and have `wpa_supplicant` to fall back on, but it would be more efficient for the firmware blob to handle it, and would allow WPA2 encrypted links on more bare metal/embedded platforms including [RISC OS](https://www.riscosopen.org/tracker/tickets/638) and gokrazy who also seem [stuck on Pi 400 WiFi](https://github.com/gokrazy/gokrazy/issues/188).

I've tried 12 different firmware blobs from various places[^1] but none appears to connect, though the function signatures hidden in the binaries do suggest something about WPA2 is supported, just not the 4-way handshake it seems.

Is it possible to request a firmware blob from the chip vendor with this enabled? Even if, for experimenting, that was at the expense of removing some other feature to save space such as the ability to be a hotspot (for example). Thanks!

[^1]: https://github.com/armbian/firmware/blob/master/brcm/ and https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/brcm and https://github.com/cobratbq/pinebook-pro/tree/main and https://github.com/LibreELEC/brcmfmac_sdio-firmware-rpi/

pelwell, 2024-07-06:
It sounds like you are essentially asking for a custom firmware for non-Linux OSes. Given his hard it is to get support for problems affecting Linux, this is realistically not going to happen.

closed 2024-07-06 by pelwell

rps102, 2024-07-06:
> It sounds like you are essentially asking for a custom firmware for non-Linux OSes

Partly that, inspired by the success I saw with the similar question for the linked BCM43430/2 issue, but also that even with Linux OSes it'd remove a bunch of chugging backwards and forwards over the (relatively slow) SDIO bus if the firmware could do it on-chip - the brcmfmac driver already supports it so "all" that's needed is the corresponding firmware.