RPi-Distro/firmware-nonfree #50: Packages kept back after yesterday's updates

XhmikosR, 2025-03-12:
I'm using a Raspberry Pi Zero 2 W with Raspberry Pi OS 64-bit (bookworm).

After yesterday's updates I saw 2 packages kept back. The other packages were updated:

```
Start-Date: 2025-03-11  21:27:43
Commandline: apt-get upgrade
Requested-By: pi (1000)
Upgrade: firmware-brcm80211:arm64 (1:20230625-2+rpt3, 1:20240709-2~bpo12+1+rpt2), firmware-atheros:arm64 (1:20230625-2+rpt3, 1:20240709-2~bpo12+1+rpt2), firmware-realtek:arm64 (1:20230625-2+rpt3, 1:20240709-2~bpo12+1+rpt2)
End-Date: 2025-03-11  21:28:25
```

```
The following packages have been kept back:
  firmware-libertas firmware-misc-nonfree
```

If I do `sudo apt install firmware-libertas firmware-misc-nonfree` then I'm getting the following which doesn't seem right?

```
pi@rpi0-2w:~ $ sudo apt install firmware-libertas firmware-misc-nonfree
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  firmware-intel-graphics firmware-intel-misc firmware-marvell-prestera firmware-mediatek firmware-nvidia-graphics
The following NEW packages will be installed:
  firmware-intel-graphics firmware-intel-misc firmware-marvell-prestera firmware-mediatek firmware-nvidia-graphics
The following packages will be upgraded:
  firmware-libertas firmware-misc-nonfree
2 upgraded, 5 newly installed, 0 to remove and 0 not upgraded.
Need to get 145 MB of archives.
After this operation, 125 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

I'm unsure if it's correct or if it's a mistake, but installing NVIDIA doesn't seem right?

/CC @XECDesign since you seem to have pushed the changes

Thanks!

XhmikosR, 2025-03-13:
I went ahead and run `sudo apt dist-upgrade`. After the new packages were installed, I purged them.

But makes me wonder if this is a mistake because it's about ~10% of the current OS size and I'm pretty sure these firmware packages are not needed for Pi.

XECDesign, 2025-03-13:
Yup, `apt upgrade` is not enough in many cases, so that's not what we recommend in [the documentation](https://www.raspberrypi.com/documentation/computers/os.html#install-updates).

Previously firmware-misc-nonfree contained all of the firmware that are now in the new packages. As you've pointed out, that's a lot of data that most people won't need, especially on a Raspberry Pi. So Debian's maintainers have split it across multiple packages, which allows the user to remove things they don't need.

I think future images will ship without the intel and nvidia graphics to save a bit of space.

XhmikosR, 2025-03-13:
Thanks for the reply!

Are the other firmware packages needed, though?

XECDesign, 2025-03-13:
I think the only one you really need is firmware-brcm80211, but it depends on what other hardware you may have plugged in.

XhmikosR, 2025-03-13:
I just have a Raspberry Pi Zero 2 W with an Ethernet USB HAT which is using Realtek firmware AFAICT. Things seem to still work fine without those extra firmware files that got downloaded after I removed them.

Thanks for the replies!